boschsensortec/BME280_SensorAPI #1: Code quality

Manu3l0us, 2015-03-05:
Seriously Bosch, that's not how it's done. 

   #define BME280_TEMP_1_2_2_8_8_0_DATA		122880
   #define BME280_PRESSURE_6_4_0_0_0_DATA		64000
   #define BME280_PRESSURE_3_1_2_5_DATA		3125
   #define BME280_PRESSURE_3_2_7_6_8_DATA		32768
   #define BME280_PRESSURE_1_0_4_8_5_7_6_DATA			1048576
   #define BME280_HEX_PRESSURE_8_0_0_0_0_0_0_0_DATA	0x80000000

ddkumar, 2015-03-13:
The supplied driver code is lacking in several respects.  As Manu310us remarks, the code is very confusing to read due to the use of elaborate #defines for all constants and shift values.  Furthermore, the BME280 datasheet provides a 64 bit pressure compensation algorithm which has no corresponding implementation in the driver, yet the datasheet recommends you use the driver code.  I have rewritten the compensation to use a more conventional format and included both 64b and 32b pressure compensations.  I would be willing to post the source, if there is interest.

vanbako, 2015-09-25:
Hello ddkumar,
I'm interested in the code.
Do you know what BME280_ONE_U8X is in the code?

Manu3l0us, 2015-09-25:
It's probably just "(uint8_t)1u". The code is/was full of such constants.

BoschSensortec, 2016-06-30:
Thanks for your input. We keep improving our code quality for each release.

closed 2016-10-25 by BoschSensortec