boschsensortec/BME280_SensorAPI #15: Sample Data?

TPLindley, 2016-09-22:
Do you have some sample data you could post? I have ported to Windows UWP IoT and am reading the data from the chip fine, and think I have the reference values as well, but the compensated values are all wrong. Would love to know what I should expect for raw temp at 70d F for example. I am getting the following values with my testing.

RawTemp 0x00000757
CompensatedTemp (UInt) 0xfffff12f
Converted Celcius -37.93
DigT1 0x0000196e
DigT2 0x00007e66
DigT3 0x00003200

Is there something I need to do before I read the compensation values from the chip? Right now, I am basically grabbing the Chip ID for verification purposes (works fine), then reading the compensation values, then getting the uncompensated temp/press/hum values.

Thanks for any help you can provide.

BoschSensortec, 2016-09-23:
Thank you for your comments.
It seems the trim values which are mentioned is not correct. 
Please provide the register dump along with API functions which are called and also the mode which you are using
It will be better if you could provide the source code which you are using to call BME280 API.
Also you can contact our FAE for further help.

TPLindley, 2016-09-23:
Here is the code, a c# class for Windows IoT Core.
using Common.Models;
using System;
using System.Collections.Generic;
using System.Threading.Tasks;
using Windows.Devices.Enumeration;
using Windows.Devices.I2c;
using Windows.UI.Xaml;

namespace Common.Devices
{
    public static class BME280Commands
    {
        #region Static
        public static byte[] Sample1 = { 0x01 };

        public static byte[] ChipID = { 0xD0 };
        public static byte[] Version = { 0xD1 };
        public static byte[] SoftReset = { 0xE0 };

        public static byte[] ControlHumidity = { 0xF2, 0x01 };
        public static byte[] Control = { 0xF4, 0x21 };
        public static byte[] Config = { 0xF5 };
        public static byte[] PressureData = { 0xF7 };
        public static byte[] PressureData1 = { 0xF8 };
        public static byte[] PressureData2 = { 0xF9 };
        public static byte[] TemperatureData = { 0xFA, 0xFB, 0xFC };
        public static byte[] TemperatureData1 = { 0xFB };
        public static byte[] TemperatureData2 = { 0xFC };
        public static byte[] HumidityData = { 0xFD };
        public static byte[] HumidityData1 = { 0xFE };
        public static byte[] HumidityData2 = { 0xFF };

        public static byte[] DIG_T1 = { 0x88 };
        public static byte[] DIG_T2 = { 0x8A };
        public static byte[] DIG_T3 = { 0x8C };

        public static byte[] DIG_P1 = { 0x8E };
        public static byte[] DIG_P2 = { 0x90 };
        public static byte[] DIG_P3 = { 0x92 };
        public static byte[] DIG_P4 = { 0x94 };
        public static byte[] DIG_P5 = { 0x96 };
        public static byte[] DIG_P6 = { 0x98 };
        public static byte[] DIG_P7 = { 0x9A };
        public static byte[] DIG_P8 = { 0x9C };
        public static byte[] DIG_P9 = { 0x9E };

        public static byte[] DIG_H1 = { 0xA1 };
        public static byte[] DIG_H2 = { 0xE1 };
        public static byte[] DIG_H3 = { 0xE3 };
        public static byte[] DIG_H4 = { 0xE4 };
        public static byte[] DIG_H5 = { 0xE5 };
        public static byte[] DIG_H6 = { 0xE7 };

        public static byte ExpectedChipId = 0x60;
        public static byte SlaveAddress = 0x77;
        public static byte SoftResetCode = 0xb6;
        #endregion
    }
    public class BME280 : CommonBase
    {
        #region Properties
        public double Celcius { get { return _compensatedTemp/100.0; } }                // Celcius
        public double Humidity { get { return _compensatedHumidity / 1024.0; } }        //% relative humidity
        public double Pressure { get { return _compensatedPressure / 100.0; } }         // Pressure in millibars
        public bool Connected { get; private set; }
        #endregion
        #region Public
        public BME280()
        {
[... 196 lines cut ...]
                else
                    return 0;
            }
            v_x1 = (((Int32)_digP9) * ((Int32)(((v_pressure >> 3) * (v_pressure >> 3)) >> 13))) >> 12;
            v_x2 = (((Int32)(v_pressure >> 2)) * ((Int32)_digP8)) >> 13;
            v_pressure = (UInt32)((Int32)v_pressure + ((v_x1 + v_x2 + _digP7) >> 4));
            return v_pressure;
        }
        private UInt32 CompensateHumidity(Int32 rawHumidity)
        {
            UInt32 v_x1_u32 = 0;

            /* calculate x1*/
            v_x1_u32 = (UInt32)(_fineTemp - ((Int32)76800));
            /* calculate x1*/
            v_x1_u32 = (UInt32)(((((rawHumidity << 14) - (((Int32)_digH4) << 20) - (((Int32)_digH5) * v_x1_u32)) + ((Int32)16384)) >> 15) * (((((((v_x1_u32 * ((Int32)_digH6)) >> 10) * (((v_x1_u32 * ((Int32)_digH3)) >> 11) + ((Int32)32768))) >> 10) + ((Int32)2097152)) * ((Int32)_digH2) + 8192) >> 14));
            v_x1_u32 = (UInt32)(v_x1_u32 - (((((v_x1_u32 >> 15) * (v_x1_u32 >> 15)) >> 7) * ((Int32)_digH1)) >> 4));
            v_x1_u32 = (v_x1_u32 < 0 ? 0 : v_x1_u32);
            v_x1_u32 = (v_x1_u32 > 419430400 ?
            419430400 : v_x1_u32);
            return (UInt32)(v_x1_u32 >> 12);
        }
        #endregion
    }
}

TPLindley, 2016-09-23:
Wondering if there is something I need to call/do before reading the config values? Thanks for the help.

TPLindley, 2016-09-26:
I added setting the config and humid/press/temp settings. Still getting the same result. Here is a snip of the changes I made to add setting these values.
            var buf1 = new byte[1];
            try
            {
                buf1[0] = 0x16;        // Set configuration to Normal Mode, standby at .5ms
                bme280Sensor.WriteRead(BME280Commands.Config, buf1);

                buf1[0] = 0x01;       // Set humidity oversampling x 1
                bme280Sensor.WriteRead(BME280Commands.ControlHumidity,buf1);

                buf1[0] = 0x87;      // Set press oversample x 16, temp oversample x 2, filter to coefficient 16
                bme280Sensor.WriteRead(BME280Commands.Control, buf1);

                bme280Sensor.WriteRead(BME280Commands.DIG_T1, buf2);
                _digT1 = (UInt32)((buf2[0] << 8) | buf2[1]);

And here are the values I am reading from the BME280:
DT1 6510, DT2 32358, DT3 12800
DP1 51087, DP2 57045, DP3 53259, DP4 44571, DP5 3840, DP6 63999, DP7 44070, DP8 2776, DP9 48400
DH1 75, DH2 27905, DH3 0, DH4 4872, DH5 2048, DH6 30
Raw Temp: 1783, Raw Press: 524288, Raw Humid: 28535
BME280: Temp: -37.97c Pressure: 1451.62 millibars Humidity: 100 %rH

TPLindley, 2016-09-26:
And in hex
DT1 0x196e, DT2 0x7e66, DT3 0x3200
DP1 0xc78f, DP2 0xded5, DP3 0xd00b, DP4 0xae1b, DP5 0xf00, DP6 0xf9ff, DP7 0xac26, DP8 0xad8, DP9 0xbd10
DH1 0x4b, DH2 0x6d01, DH3 0x0, DH4 0x1308, DH5 0x800, DH6 0x1e
Raw Temp: 0x6f6, Raw Press: 0x80000, Raw Humid: 0x6f6f

BME280: Temp: -37.97c Pressure: 1451.62 millibars Humidity: 100 %rH

BoschSensortec, 2016-10-25:
@TPLindley , The Endian format is incorrect. For example, DT1 should be used as 0x6e19.

closed 2017-07-14 by BoschSensortec