boschsensortec/BME280_SensorAPI #16: BME280 on Windows 10 IoT Core

TPLindley, 2016-09-27:
Created this second post as have not received any responses on code submission. I am able to reset the chip, read the chip ID, get the trimming values and read the temp/press/hum values (burst mode). However, the compensated values are not correct. I am hoping you can supply some sample values so I can see where the problem lies. Here are the values I am getting now.

DT1 0x196e, DT2 0x7e66, DT3 0x3200
DP1 0xc78f, DP2 0xded5, DP3 0xd00b, DP4 0xae1b, DP5 0xf00, DP6 0xf9ff, DP7 0xac26, DP8 0xad8, DP9 0xbd10
DH1 0x4b, DH2 0x6d01, DH3 0x0, DH4 0x1308, DH5 0x800, DH6 0x1e
Raw Temp: 0x7fd10, Raw Press: 0x581bc, Raw Humid: 0x6cca

Compensated values here:
BME280: Temp: 187.36c Pressure: 1344.55 millibars Humidity: 100 %rH

I am sure I am just doing something dumb, just need a little nudge in the right direction. Thanks.

TPLindley, 2016-09-27:
Code here:
    public static class BME280Commands
    {
        #region Static
        public static byte[] Sample1 = { 0x01 };

        public static byte[] ChipID = { 0xD0 };
        public static byte[] Version = { 0xD1 };
        public static byte[] SoftReset = { 0xE0, 0xB6 };

        public static byte[] ControlHumidity = { 0xF2, 0x04 };  // second parm was 0x01
        public static byte[] Status = { 0xF3 };
        public static byte[] Control = { 0xF4, 0x91 };          // second parm was 0x87
        public static byte[] Control1 = { 0xf4, 0x3f };
        public static byte[] Config = { 0xF5, 0x16 };
        public static byte[] PressureData = { 0xF7 };
        public static byte[] TemperatureData = { 0xFA };
        public static byte[] HumidityData = { 0xFD };

        public static byte[] DIG_T1 = { 0x88 };
        public static byte[] DIG_T2 = { 0x8A };
        public static byte[] DIG_T3 = { 0x8C };

        public static byte[] DIG_P1 = { 0x8E };
        public static byte[] DIG_P2 = { 0x90 };
        public static byte[] DIG_P3 = { 0x92 };
        public static byte[] DIG_P4 = { 0x94 };
        public static byte[] DIG_P5 = { 0x96 };
        public static byte[] DIG_P6 = { 0x98 };
        public static byte[] DIG_P7 = { 0x9A };
        public static byte[] DIG_P8 = { 0x9C };
        public static byte[] DIG_P9 = { 0x9E };

        public static byte[] DIG_H1 = { 0xA1 };
        public static byte[] DIG_H2 = { 0xE1 };
        public static byte[] DIG_H3 = { 0xE3 };
        public static byte[] DIG_H4 = { 0xE4 };
        public static byte[] DIG_H5 = { 0xE5 };
        public static byte[] DIG_H6 = { 0xE7 };

        public static byte ExpectedChipId = 0x60;
        public static byte SlaveAddress = 0x77;
        public static byte SoftResetCode = 0xb6;

TPLindley, 2016-09-27:
var data = new byte[1];
            bme280Sensor.WriteRead(BME280Commands.SoftReset, data);

            bme280Sensor.WriteRead(BME280Commands.ChipID, Id);
            if (Id[0] == BME280Commands.ExpectedChipId)
            {  // How I read the trimming values
                bme280Sensor.WriteRead(BME280Commands.DIG_T1, buf2);
                _digT1 = (UInt32)((buf2[0] << 8) | buf2[1]);
                bme280Sensor.WriteRead(BME280Commands.DIG_T2, buf2);
                _digT2 = ((buf2[0] << 8) | buf2[1]);
                bme280Sensor.WriteRead(BME280Commands.DIG_T3, buf2);
                _digT3 = ((buf2[0] << 8) | buf2[1]);

// How I read the actual values
                bme280Sensor.Write(BME280Commands.ControlHumidity);
                bme280Sensor.Write(BME280Commands.Control);

                var sleepTime = 0.00125 + 0.0023 * (1 << 4);
                sleepTime = sleepTime + 0.0023 * (1 << 4) + 0.000575;
                sleepTime = (sleepTime + 0.0023 * (1 << 4) + 0.000575)*1000;
                Task.Delay((int)sleepTime);

                bme280Sensor.WriteRead(BME280Commands.PressureData, buf8);
                _rawPress = (buf8[0] << 16 | buf8[1] << 8 | buf8[2]) >> 4;
                _rawTemp = (buf8[3] << 16 | buf8[4] << 8 | buf8[5]) >> 4;
                _rawHumid = (buf8[6] << 8) | buf8[7];

BoschSensortec, 2016-10-25:
@TPLindley ,
Can you share information on how to replicate your work setup so we can incorporate it in our development process.

Thank you.  And we apologize for the delay.

BoschSensortec, 2017-07-14:
Closing due to inactivity.

closed 2017-07-14 by BoschSensortec