boschsensortec/BME280_SensorAPI #17: I2C_BUFFER_LEN too small--stack overflow in BME280_I2C_bus_read

DavidPletcher, 2017-01-18:
In the BME280_I2C_bus_read function, only 8 bytes (I2C_BUFFER_LEN) are allocated for the I2C read/write buffer. The bme280_get_calib_param function reads 26 bytes into that buffer, overflowing the stack and overwriting the instruction pointer such that the program never returns from bme280_get_calib_param. This is a serious bug. Please fix.

#define I2C_BUFFER_LEN 8
...
u8 array[I2C_BUFFER_LEN] = {BME280_INIT_VALUE};
...
#define	BME280_CALIB_DATA_SIZE			(26)
...
com_rslt = p_bme280->BME280_BUS_READ_FUNC(
p_bme280->dev_addr,
BME280_TEMPERATURE_CALIB_DIG_T1_LSB_REG,
a_data_u8,
BME280_PRESSURE_TEMPERATURE_CALIB_DATA_LENGTH);

BoschSensortec, 2017-01-20:
@DavidPletcher,

This will be corrected to the earliest.

amki, 2017-06-03:
The same is true für SPI_BUFFER_SIZE. It is set to 5 but the calibration data is still 26 long and overflowing the buffer.

BoschSensortec, 2017-06-23:
@amki and @DavidPletcher ,

Please find the updated driver for the [BME280](https://github.com/BoschSensortec/BME280_driver).

closed 2017-06-23 by BoschSensortec