boschsensortec/BME280_SensorAPI #20: BME280 with WICED43907

AttilaMak, 2017-06-26:
Hi All,

I have a Bosch BME280 sensor. I am working with Cypress's WICED platform at the moment.
With SPI. I'm tring to get the measurement results from the sensor, but it gaves me 0 as temperature.
I I1m ask back the coefficient the answer:
CHIP ID: 60
_bme280_calib.dig_T1 val: 6F0D
_bme280_calib.dig_T2 val: 6844
_bme280_calib.dig_T3 val: 32
_bme280_calib.dig_P1 val: 91FF
_bme280_calib.dig_P2 val: FFFFD6FF
_bme280_calib.dig_P3 val: BD0
_bme280_calib.dig_P4 val: 1465
_bme280_calib.dig_P5 val: 35
_bme280_calib.dig_P6 val: FFFFFFFF
_bme280_calib.dig_P7 val: 26AC
_bme280_calib.dig_P88 val: FFFFD8FF
_bme280_calib.dig_P9 val: 10BD
_bme280_calib.dig_H1 val: 4B
_bme280_calib.dig_H2 val: 166
_bme280_calib.dig_H3 val: 0
_bme280_calib.dig_H4 val: 148
_bme280_calib.dig_H5 val: 0
_bme280_calib.dig_H6 val: 1E

So it seems there is communication on SPI bus. Maybe calibration mistake?
could somebody help me please?

Many thanks!
[Bosch_Sensorboards.zip](https://github.com/BoschSensortec/BME280_driver/files/1101310/Bosch_Sensorboards.zip)

BoschSensortec, 2017-06-26:
@AttilaMak ,

The calibration values seem plausible. I am not sure about the architecture of the processor that you are using, but typically this could be an issue related endianess. 

It is recommended to use the API provided where you only have to link at run-time the read, write and delay functions in order to use the complete set of APIs to operate the sensor. 

In case you would like to write your own driver, then refer to the integer temperature calculation at line 1011 in [bme280.c](https://github.com/BoschSensortec/BME280_driver/blob/master/bme280.c) . Can you also share the raw value read from the sensor. It could be possible that the sensor is not in an active state.

AttilaMak, 2017-06-27:
Hi,

Thanks for the answer. You are right I forgot to mention that I checked with the oscilloscope and the answer is 0. So All bits are zero in the answer.  – So that is why I’m thinking about my wrong initialization.

Many thanks in advance!

További szép napot! / Have a nice day!
_________________________________
Attila Mák
Application Engineer – ESC
Arrow Electronics
H-1138, Budapest
Vaci Str 140.
Hungary

Phone: +3612887372
E-mail:  amak@arroweurope.com<mailto:amak@arroweurope.com>
www.arrow.com<http://www.arrow.com/> / www.arroweurope.com<http://www.arroweurope.com/>
[Arrow_Global_Logo]
[http://farm1.static.flickr.com/205/492481199_733a0b5929.jpg]    [http://www.concordia.edu/sitefiles/w3/Univservices/web-banner-thinkB4Uprint.png?c=19094]

[cid:image003.png@01CFE2EF.F8A8B1E0]

closed 2017-07-14 by BoschSensortec