boschsensortec/BME280_SensorAPI #21: I got an ERROR in IAR with Version: 3.2.0

jiekechoo, 2017-06-27:
Error[Pe028]: expression must have a constant value F:\code\STM32F103BME280 - boss\DRIVER\src\BME280.c 448

int8_t bme280_set_regs(uint8_t *reg_addr, const uint8_t *reg_data, uint8_t len, const struct bme280_dev *dev)
{
uint8_t temp_buff[len * 2];
}

jiekechoo, 2017-06-28:
[image]
and the pressure is abnormal.

BoschSensortec, 2017-07-10:
@jiekechoo ,

It seems that your compiler doesn't support dynamic length arrays. We will update this shortly to support cases where only static length arrays are supported.

jiekechoo, 2017-07-11:
@BoschSensortec 
I found the solution, Allow VLA in IAR, fixed.

And, the pressure indicators seem unstable.

GhostRiderLi, 2017-07-13:
there there two errors about shift bits in line 853 and 859 of  bme280.c

tkrahn108, 2017-07-13:
@GhostRiderLi What is the correct bit shifting?

muchajan, 2017-07-13:
@tkrahn108 I think, according to the documentation, there should be right shift by 4, not the left shift

tkrahn108, 2017-07-13:
@muchajan That was also my idea but it doesn't solve my issue #22.

muchajan, 2017-07-13:
@tkrahn108 Did you check the compensation functions? They seems to differ with the code in documentation

tkrahn108, 2017-07-13:
@muchajan Before I used this API I had written my own driver with the compensation function provided in the data sheet. But I got the same values. Therefore, I think the unconsistency is not due to the compensation function.

MMMUTU, 2017-07-20:
Hello gentlemen, i have one silly question not even worthy a new thread, what am i supposed to fill in those strings:
dev.read = user_i2c_read;
dev.write = user_i2c_write;
dev.delay_ms = user_delay_ms;
I have i2c bme280 connection with STM32F429ZI. Pins: PA8 is SCL, PC9 is SDA.
Should i fill anything else anywhere?

BoschSensortec, 2017-07-20:
@MMMUTU ,

We have now added function templates in the README for the user_i2c_read, user_i2c_write and user_delay_ms and describing what each of these functions should do in the latest release.

BoschSensortec, 2017-07-20:
@GhostRiderLi ,

Thank you for your prompt feedback. Do find your comments fixed in the latest release.

BoschSensortec, 2017-07-20:
@jiekechoo ,

Error regarding variable length arrays as been fixed by using a fixed length array for the write. Please close the issue if you do not have any issues related to the v3.3.0 of the driver

jiekechoo, 2017-07-22:
[image]

@BoschSensortec ,
I updated to version 3.3.0, the pressure seems OK. Thank you!

closed 2017-07-22 by jiekechoo