boschsensortec/BME280_SensorAPI #22: Inconsistency of sensor values

tkrahn108, 2017-07-12:
I'm working on a project with actually 8 BME280 on 8 different PCBs with the nRF51822 as microcontroller. For communication the I2C interface is used. I am using the provided API as well as the initialization and forced mode readout function from the readme (I only changed the oversampling rate to 1x), but the sensor values are very inconsistent. 

In the following table the values for the 8 different sensors are listed. I added reference values from a home weather station in the last column.

  | 1 | 2 | 3 | 4 | 5 | 6 | 7 | 8 | weather station
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
Humidity | 62,08 | 63,4 | 72,8 | 94,6 | 89 |  63,8 | 97,7 | 63,6 | 59
Pressure | 996,12 | 995,8 | 996,37 | 990,91 | 994,2 | 990,34 | 1001,9 | 994,5 | 1004,3
Temperature | 25,73 | 26,4 | 26,3 | 25,76 | 26,2 | 25,9 | 25,97 | 26,35 | 24,2

Do you have any ideas what's going wrong?

BoschSensortec, 2017-07-14:
@tkrahn108 ,

What is the time interval between each of the measurements?

 A fix based on #21 is in progress and will be released soon. That might explain the deviation in the pressure and temperature signals. regarding the humidity signal, it could be data from a real change in humidity near the sensor.

tkrahn108, 2017-07-14:
@BoschSensortec 

At the moment the interval between each measurement is set to 5 seconds. 

Hopefully it will help. But I already tried to correct the base shift and it didn't solve this problem. Regarding humidity, I don't think it is a real change in humidity as the sensors are lying side by side. This cannot explain a deviation of about 30 %, can it?

BoschSensortec, 2017-07-17:
@tkrahn108 ,

Is it possible for you to get a bus transaction log using a logic analyzer, like the Saleae for example? 

Also, I assume you are using the newest version of the API.

tkrahn108, 2017-07-18:
@BoschSensortec ,
at the moment I have no logic analyzer but I will try to get one. 

Yes, I'm using the newest API version 3.2.0.

Maybe some additional information could help you: I noticed that at least the pressure and temperature values vary linearly from the weather station's values for device 1 (Pressure: ~-8 hPa, Temperature: ~+1,5°C ). Therefore, I suppose that there is something wrong with the compensation function or/and the trimming parameters.

BoschSensortec, 2017-07-20:
@tkrahn108 ,

Can you please check with the latest release v3.3.0 if the problem still persists. Also, you can create a communication log by printing the data passed to and from the read/write functions. Remember to print before the SPI/I2C write transaction inside the wrapper and after the SPI/I2C transaction during read.

tkrahn108, 2017-07-20:
@BoschSensortec ,
yes, the problem still persists with v3.3.0. 

Here is the communication log for device 1. It includes initialization, configuration as in the `stream_sensor_data_normal_mode` of the README and the first five data readouts.

```
bme_280_read
dev_id: 76	 reg_addr: D0	 len: 1
data: 60 	

bme_280_write
dev_id: 76	 reg_addr: E0	 len: 1
data: B6 	

bme_280_read
dev_id: 76	 reg_addr: 88	 len: 26
data: AC 	6D 	B2 	66 	32 	00 	4C 	91 	E2 	D5 	D0 	0B 	FC 	1B 	E4 	FF 	F9 	FF 	AC 	26 	0A 	D8 	BD 	10 	00 	4B 	

bme_280_read
dev_id: 76	 reg_addr: E1	 len: 7
data: 6D 	01 	00 	13 	07 	00 	1E 	

bme_280_read
dev_id: 76	 reg_addr: F4	 len: 1
data: 00 	

bme_280_write
dev_id: 76	 reg_addr: F2	 len: 1
data: 01 	

bme_280_read
dev_id: 76	 reg_addr: F4	 len: 1
data: 00 	

bme_280_write
dev_id: 76	 reg_addr: F4	 len: 1
data: 00 	

bme_280_read
dev_id: 76	 reg_addr: F4	 len: 1
data: 00 	

bme_280_write
dev_id: 76	 reg_addr: F4	 len: 1
data: 54 	

bme_280_read
dev_id: 76	 reg_addr: F5	 len: 1
data: 00 	

bme_280_write
dev_id: 76	 reg_addr: F5	 len: 1
data: 30 	

bme_280_read
dev_id: 76	 reg_addr: F4	 len: 1
data: 54 	

bme_280_read
dev_id: 76	 reg_addr: F4	 len: 1
data: 54 	

bme_280_write
dev_id: 76	 reg_addr: F4	 len: 1
data: 57 	

bme_280_read
dev_id: 76	 reg_addr: F7	 len: 8
data: 53 	10 	70 	82 	17 	30 	7D 	66 	
2621	99872	70640

bme_280_read
dev_id: 76	 reg_addr: F7	 len: 8
data: 53 	12 	70 	82 	1E 	70 	7D 	63 	
2625	99872	70635

bme_280_read
dev_id: 76	 reg_addr: F7	 len: 8
data: 53 	13 	40 	82 	21 	C0 	7D 	66 	
2627	99873	70649

bme_280_read
dev_id: 76	 reg_addr: F7	 len: 8
data: 53 	13 	A0 	82 	23 	60 	7D 	6C 	
2627	99872	70684

bme_280_read
dev_id: 76	 reg_addr: F7	 len: 8
data: 53 	13 	D0 	82 	24 	80 	7D 	6F 	
2628	99877	70708
```

The corresponding values of the weather station (if you are interested in) were 25,0°C, 1006,8 hPa and 63%.

BoschSensortec, 2017-07-21:
@tkrahn108 ,

Would it be possible for you to share more details about your hardware setup?

In addition, you could perform an overnight experiment using an air-tight box to achieve a constant humidity of 75% using a [saturated table salt (sodium chloride) solution](http://www.omega.com/temperature/z/pdf/z103.pdf) and measuring the humidity and temperature overnight.

tkrahn108, 2017-07-21:
@BoschSensortec ,
I am allowed to provide some hopefully relevant details of the schematic. 
[image]
[image]

Regarding the experiment, I will try to perform the experiment within the next few days.

tkrahn108, 2017-07-24:
@BoschSensortec ,

I have performed the experiment last night with three devices (1, 4, 6). I obtained the following data
[BME280_#1.xlsx](https://github.com/BoschSensortec/BME280_driver/files/1168868/BME280_.1.xlsx)
[BME280_#4.xlsx](https://github.com/BoschSensortec/BME280_driver/files/1168869/BME280_.4.xlsx)
[BME280_#6.xlsx](https://github.com/BoschSensortec/BME280_driver/files/1168870/BME280_.6.xlsx)

Hopefully, I achieved a constant humidity of 75% within the air-tight box. I have controlled with the weather station and get a humidity of 73% this morning. But it is not excluded that there is some measurement error for this device, too. Otherwise, I don't know whether the box was really air-tight. Despite this I hope this data will be useful.

One more explanatory note for the data of device 4. The humidity was constant since 21:56:51 until I stopped the experiment this morning. 

In addition, I want to provide some diagrams for a first view:

**Device 1**

[image]
[image]
[image]

**Device 4**
[image]
[image]
[image]

**Device 6**
[image]
[image]
[image]

BoschSensortec, 2017-07-25:
@tkrahn108 ,

We will investigate this and get back to you shortly.

BoschSensortec, 2017-07-28:
@tkrahn108 ,

We setup a test with 2 sensors controlled by an Cortex M0+micro-controller and observed that the humidity was within the accuracy tolerance of 3%, 72% - 78%. The test was conducted in an air tight box and the humidity was saturated to 75% with the help of a saturated Sodium Chloride solution.

Can you comment on the compiler you are currently using for the nrf51822 and related compile options?

tkrahn108, 2017-07-28:
@BoschSensortec ,

I am using the ARM compiler V5.06 update 4 (build 422) with the follwing control string:

```
-c --cpu Cortex-M0 -D__EVAL -D__MICROLIB -g -O3 --apcs=interwork --split_sections 
-I ../../../config 
-I ../../../../../bsp 
-I ../../../../../../components/toolchain 
-I ../../../../../../components/ble/ble_advertising 
-I ../../../../../../components/ble/common 
-I ../../../../../../components/ble/device_manager 
-I ../../../../../../components/drivers_nrf/common 
-I ../../../../../../components/drivers_nrf/delay 
-I ../../../../../../components/drivers_nrf/gpiote 
-I ../../../../../../components/drivers_nrf/config 
-I ../../../../../../components/drivers_nrf/hal 
-I ../../../../../../components/drivers_nrf/pstorage 
-I ../../../../../../components/drivers_nrf/uart 
-I ../../../../../../components/libraries/button 
-I ../../../../../../components/libraries/timer 
-I ../../../../../../components/libraries/trace 
-I ../../../../../../components/libraries/uart 
-I ../../../../../../components/libraries/util 
-I ../../../../../../components/softdevice/common/softdevice_handler 
-I ../../../../../../components/softdevice/s110/headers 
-I C:/Keil_v5/ARM/Pack/NordicSemiconductor/RTT 
-I ../arm5_no_packs -I ../../../../../../components/drivers_nrf/twi_master 
--c99
-IC:/Temp/.../pca10028/s110/arm5_no_packs/RTE/_nrf51422_xxac_s110
-IC:/Keil_v5/ARM/PACK/ARM/CMSIS/4.5.0/CMSIS/Include
-IC:/Keil_v5/ARM/PACK/NordicSemiconductor/nRF_DeviceFamilyPack/8.2.0/Device/Include
-IC:/Keil_v5/ARM/PACK/NordicSemiconductor/nRF_DeviceFamilyPack/8.9.0/Device/Include
-D__UVISION_VERSION="522" -D_RTE_ -DNRF51 -DBLE_STACK_SUPPORT_REQD -DS110 -DBOARD_PCA10028 -DSOFTDEVICE_PRESENT -DNRF51 -DSWI_DISABLE0
-o .\_build\*.o --omf_browse .\_build\*.crf --depend .\_build\*.d
```

BoschSensortec, 2017-08-04:
@tkrahn108 ,

We tried replicating it with our sensors but do not see the sensor to be out of spec. It might be that due to storage conditions or other factors, the sensor may need reconditioning. You can find information related to that in the [Handling, Soldering, & Mounting Instructions document](https://ae-bst.resource.bosch.com/media/_tech/media/application_notes/BST-BME280-HS001-06.pdf). Try the 2nd procedure under Chapter 3.3.

marriotb, 2017-08-10:
@tkrahn108 
It's really worth while looking at your physical setup, too, in particular in relation to airflow across the devices.

I have a bunch of sensors that were showing similar discrepancies in readings to yours. Three sensors sitting 5cm away from each other gave readings that were across a range of 20 percentage points - eg 50%, 60% and 70% from each of the three sensors at one time.

I decided to try comparing them against a known standard. Not having access to anything more sophisticated, I set up a wet/dry bulb thermometer system using some DS18B20 devices. I had previously tested these particular ones and knew I could rely on them to be within 0.1C of each other.

What I discovered was that, provided I had a fan blowing air across the group of sensors, the BME280s were within a couple of percentage points of each other and of the humidity calculated from the wet and dry readings (plus pressure, which was reliably supplied by the BME280s). A steady airflow is a requirement of the wet/dry bulb system to get sufficient evaporation from the surface of the wet bulb thermometer. I just used a desk fan on slow-speed setting (barely enough to ruffle papers)

If I turned off the fan, the devices quickly returned to their previous behaviour - a wide spread of readings. So it's clear that they need airflow across them to get reliable humidity readings. (Not too much, or the pressure readings will suffer:-)

Regards, Brian

tkrahn108, 2017-08-11:
@marriotb 
Thank you for sharing your experiences with the BME280. 

Do your devices have a casing? My devices don't have a housing at the moment.

tkrahn108, 2017-08-16:
@BoschSensortec ,
I performed the reconditioning procedure (I couldn't control humidity during dry-baking, hopefully it was <5 %), but the humidity sensors still have an offset. It seems that the offset for device 1 is even larger. Afterwards, I performed the test with the sodium chloride solution and the air-tight box with 3 devices again. I obtained the following humidity data:
**Device 1**
[image]
**Device 3**
[image]
**Device 4**
[image]

The temperature seems to be more accurate now (within ~0.2 °C for three devices), approximately matching the weather station's data.

marriotb, 2017-08-16:
@tkrahn108 
My devices aren't in any case at the moment. I'm still to recondition them, but I'm hoping that will bring the readings closer (as it did with yours). 

The humidity in a normal domestic oven will be well under 5% at the dry-baking temperature. The absolute humidity at anything like normal ambient is such that when you raise that air to 100C or more, the relative humidity is extremely low. (I looked it up in my old pre-SI steam tables and  confirmed my expectation that that would be the case.)

BoschSensortec, 2017-08-17:
@tkrahn108 & @marriotb ,

This is a unique case. Could you please contact us via the [contact form](https://www.bosch-sensortec.com/bst/contact/contact_form/contact_form) so we can resolve the issue sooner.

Thank you.

BoschSensortec, 2017-08-17:
@tkrahn108 & @marriotb ,

In the meantime, can you post the register values for registers 0x83, 0x84, 0x85 and 0x86?

tkrahn108, 2017-08-17:
0x83: 19 	 0x84: 89 	 0x85: 2D 	 0x86: F6
0x83: 1A 	 0x84: 3B 	 0x85: 2D 	 0x86: F6
0x83: 49 	 0x84: 67 	 0x85: 76 	 0x86: CA
0x83: 49 	 0x84: A0 	 0x85: 76 	 0x86: CA
0x83: 49 	 0x84: 17 	 0x85: 76 	 0x86: CA
0x83: 49 	 0x84: 1B 	 0x85: 76 	 0x86: CA
0x83: 49 	 0x84: A2 	 0x85: 76 	 0x86: CA

BoschSensortec, 2017-08-18:
@tkrahn108 ,

We would also need the lot numbers to trace the specific part. Information can be found on page 43 of the [datasheet](https://ae-bst.resource.bosch.com/media/_tech/media/datasheets/BST-BME280_DS001-11.pdf).

tkrahn108, 2017-08-18:
055 for 2 devices and 095 for the other devices

peter100191, 2017-08-18:
I am having an almost identical problem between 5 prototype sensors. Pressure and temperature are reading correctly. Humidity is ridiculously out of spec - reads 40% LOW. Already performed reconditioning procedure in a Thermotron environmental chamber (the chamber is < 5yrs old, maintained by a university... Hopefully no problems there). Is worse now than before reconditioning.

Will read 0.00% when below 40%. Gets worse with temperature increase. At 45C and 70% humidity in a Thermotron environmental chamber, the device reads 44.5C and 0.00%RH. If I decrease temperature to 30C and increase humidity to 80%, the device reads 30%. 

Batch of BMEs ordered thru Digikey within the last 8 weeks. Could dig up batch number.
Using NRF52 series MCU.