boschsensortec/BME280_SensorAPI #23: Multiply BMP280 sensors over I2C bus

milan-de, 2017-07-13:
Hi together,

is there any possibility to have 2 or more BMP280 sensor in parallel connected over I2C bus, so to be able to masure on two different points temp/hum at the same time ? If yes, how to set different address to different sensors ? It seems that all are having address 0x76 per default.

regards,
Milan

muchajan, 2017-07-13:
@milan-de From datasheet:
"The 7-bit device address is 111011x. The 6 MSB bits are fixed. The last bit is changeable by 
SDO value and can be changed during operation. Connecting SDO to GND results in slave 
address 1110110 (0x76); connection it to VDDIO results in slave address 1110111 (0x77), which 
is the same as BMP280’s I²C address. The SDO pin cannot be left floating; if left floating, the 
I²C address will be undefined."

BoschSensortec, 2017-07-14:
@milan-de ,

As quoted by @muchajan , There are only 2 fixed addresses possible. 

That being said, the BMP280's address can be changed during operation by toggling the SDO line. You could hence use it as a SPI style chip select and disable sensors by putting them on a secondary address, 0x77 and the device you want to communicate to at 0x76 with a help of a GPIO.

BoschSensortec, 2017-07-14:
@milan-de ,

Close this issue if this solution solves your query.

milan-de, 2017-07-14:
Since I have I2C chip/breakout bord with only 4pins Vdd/GND SDA/SCL how I
can toggle SDO line ?

Regards,
Milan

milan-de, 2017-07-15:
Thanks for support, now I can close it :-)

closed 2017-07-15 by milan-de