boschsensortec/BME280_SensorAPI #27: Values of the compensated temperature, pressure and humidity

leonf08, 2017-09-27:
Hello.
My issue relates to the values of the compensated sensor data. What are the limits of temperature, pressure and humidity after compensation? Why is the temperature maximum of 8500 and minimum of -4000 instead of 85 and -40 respectively, for instance?

BoschSensortec, 2017-09-28:
@leonf08 ,

It seems that the relevant comments are missing. Since it is in integer, it is scaled by 100.

leonf08, 2017-09-28:
So, before output on lcd, do I have to devide the measured readings by 100?

BoschSensortec, 2017-09-28:
@leonf08 ,

Yes, you will need to divide it by 100. Also, if you want the floating point precision, convert it to float and then divide it by 100 like the example below.
``` c
uint16_t humidity_int = 4550;
float humidity = (float)humidity_int / 100.0f;
```

leonf08, 2017-09-29:
Thank you for your assistance.

closed 2017-09-29 by leonf08