boschsensortec/BME280_SensorAPI #29: Sensors always return the same, wrong values

yankee14, 2017-11-02:
I am using this library with an ARM Cortex-M3 LPCXpresso 1769 development board. I am able to communicate with the sensor over the I2C bus, and do simple checks like verify the hardware ID. Using the code examples in this library's README.md, I apparently got some sensor data, but it never changes.

On every loop iteration that polls the sensors, the exact same values are printed every time, regardless of float, 32-bit, or 64-bit modes. This happens even when placing the sensor in my freezer or refrigerator.

Repeated results with floats enabled:

> Temperature, Pressure, Humidity
> 21.95, 66828.97, 72.69

Repeated results with 64-bit ints enabled:

> Temperature, Pressure, Humidity
> 2195, 6682893, 74433

Repeated results with 32-bit ints enabled:

> Temperature, Pressure, Humidity
> 2195, 66831, 74433

Why are the same values returned every time?

[My code repo using this library](https://github.com/yankee14/drone_sensor_array).

BoschSensortec, 2017-11-02:
@yankee14 ,

Which mode are you operating the sensor in? 

- Forced mode requires you to send a command to perform a measurement, and after a period of time, the measurement is ready. 

- Normal mode will perform measurements periodically depending on the standby time set.

Getting repeatedly the same measurements happens if you have triggered a forced measurement only once.

yankee14, 2017-11-02:
@BoschSensortec Thanks for the reply. I am using the normal mode code:

https://github.com/yankee14/drone_sensor_array/blob/master/src/drone_sensor_array.c#L179
https://github.com/yankee14/drone_sensor_array/blob/master/src/drone_sensor_array.c#L234

BoschSensortec, 2017-11-02:
@yankee14 ,

The main function should look like,

``` c
void handle_bme280_error(int8_t _rslt)
{
    if (_rslt < BME280_OK) {
        printf("BME280 Error code: %d", _rslt);
        for(;;); /* Trap if error */
    } else if (_rslt > BME280_OK) {
        printf("BME280 Warning code: %d", _rslt);
    }
}

int main(void)
{
    init();
    struct bme280_dev dev;
    int8_t rslt = BME280_OK;

    dev.dev_id = BME280_I2C_ADDR_SEC;
    dev.intf = BME280_I2C_INTF;
    dev.read = user_i2c_read;
    dev.write = user_i2c_write;
    dev.delay_ms = user_delay_ms;

    rslt = bme280_init(&dev);
    handle_bme280_error(rslt);
    
    rslt = stream_sensor_data_normal_mode(&dev);
    handle_bme280_error(rslt);

    for(;;) {
        
        rslt = bme280_get_sensor_data(BME280_ALL, &comp_data, dev);
        handle_bme280_error(rslt);
        print_sensor_data(&comp_data);
        
        testCCS811();

        _delay_ms(1000 * 5); // pause microcontroller
    }

    return 0 ;
}
```

yankee14, 2017-11-02:
@BoschSensortec Thanks. There were some issues getting that to compile.

1. No `struct bme280_data comp_data;` was declared in main context, so I made one.
2. You said: `rslt = bme280_get_sensor_data(BME280_ALL, &comp_data, dev);`
But the call to `bme280_get_sensor_data()` maybe should look like this? `rslt = bme280_get_sensor_data(BME280_ALL, &comp_data, &dev); // note passing dev address`
3. If anyone tried to run the code you posted using your libraries the way they are written, I think they would get stuck in the while(1) loop on the first call to `stream_sensor_data_normal_mode()` I have that loop removed in that function.

After I resolved that, I still have exactly the same problem. **The sensor data still does not change, and your `handle_bme280_error()` never prints anything because `rslt` is never anything besides `BME2890_OK`.** Thank you for your help so far!

```
int main(void)
{
    init();
    struct bme280_dev dev;
    struct bme280_data comp_data; //
    int8_t rslt = BME280_OK;

    dev.dev_id = BME280_I2C_ADDR_SEC;
    dev.intf = BME280_I2C_INTF;
    dev.read = user_i2c_read;
    dev.write = user_i2c_write;
    dev.delay_ms = user_delay_ms;

    rslt = bme280_init(&dev);
    handle_bme280_error(rslt);

    rslt = stream_sensor_data_normal_mode(&dev);
    handle_bme280_error(rslt);

    for(;;) {
        rslt = bme280_get_sensor_data(BME280_ALL, &comp_data, &dev); //
        handle_bme280_error(rslt);
        print_sensor_data(&comp_data);

        testCCS811();

        _delay_ms(1000 * 5); // pause microcontroller
    }

    return 0 ;
}
```

BoschSensortec, 2017-11-02:
@yankee14 ,

Yes, the typos were overlooked. Anyway, does the handle_bme280_error print anything if the sensor if physically disconnected?

Also, can you provide a brief log with the values passed to the user_i2c_read and user_i2c_write functions? Basically a printf that looks like
``` c
/** Insert this after the read code */
printf("RD: %d, %d", reg_addr, len);
for(uint16_t idx = 0; idx < len; idx++)
    printf(" 0x%x", reg_data[idx]);
printf("\r\n");
```

``` c
/** Insert this before the write code */
printf("WR: %d, %d", reg_addr, len);
for(uint16_t idx = 0; idx < len; idx++)
    printf(" 0x%x", reg_data[idx]);
printf("\r\n");
```

yankee14, 2017-11-02:
@BoschSensortec I commented out all the code for other devices on the I2C bus, except for the BME280. So the only communication should be between microcontroller and BME280.

---

First, I tested the error handle function by yanking the SCL wire, only while executing your bme280_init() function on [this particular line](https://github.com/yankee14/drone_sensor_array/blob/7472ce59d5e8fa75b0a7f9c841d19d9f1f74e478/src/drone_sensor_array.c#L242).

Stepping through JTAG, the bme280_init() function tries several times to verify the device ID, but predictably failed and returned the enum value *BME280_E_DEV_NOT_FOUND*, which has the signed char value -2. This passed into the error handle code you gave me earlier and printed:

`BME280 Error code: -2`

---

I inserted your printf statements in (hopefully) the appropriate places in the `user_i2c_*()` functions that get called by pointer from `bme280_dev` struct.
[Read](https://github.com/yankee14/drone_sensor_array/blob/5d1ab9f32cb9b9ec88db85cf0ff885423e0a0a3e/src/drone_sensor_array.c#L139).
[Write](https://github.com/yankee14/drone_sensor_array/blob/5d1ab9f32cb9b9ec88db85cf0ff885423e0a0a3e/src/drone_sensor_array.c#L152).

Console after running [the init line](https://github.com/yankee14/drone_sensor_array/blob/5d1ab9f32cb9b9ec88db85cf0ff885423e0a0a3e/src/drone_sensor_array.c#L256):
```
RD: 208, 1 0x60
WR: 224, 1 0xb6
RD: 136, 26 0x29 0x6f 0x45 0x68 0x32 0x0 0x2c 0x92 0xe2 0xd5 0xd0 0xb 0xa2 0x1d 0xc9 0xff 0xf9 0xff 0xac 0x26 0xa 0xd8 0xbd 0x10 0x0 0x4b
RD: 225, 7 0x6e 0x1 0x0 0x13 0x24 0x3 0x1e
```
Then the [normal mode stream() line](https://github.com/yankee14/drone_sensor_array/blob/5d1ab9f32cb9b9ec88db85cf0ff885423e0a0a3e/src/drone_sensor_array.c#L259):
```
RD: 244, 1 0x0
WR: 242, 1 0x1
RD: 244, 1 0x0
WR: 244, 1 0x0
RD: 244, 1 0x0
WR: 244, 1 0x54
RD: 245, 1 0x0
WR: 245, 1 0x30
RD: 244, 1 0x0
RD: 244, 1 0x0
WR: 244, 1 0x3
Temperature, Pressure, Humidity
RD: 247, 8 0x80 0x0 0x0 0x80 0x0 0x0 0x80 0x0
21.95, 66828.97, 72.69
```
Once inside the [loop](https://github.com/yankee14/drone_sensor_array/blob/5d1ab9f32cb9b9ec88db85cf0ff885423e0a0a3e/src/drone_sensor_array.c#L263), the first call to `bme280_get_sensor_data()`:
```
RD: 247, 8 0x80 0x0 0x0 0x80 0x0 0x0 0x80 0x0
```
[Passing the newly updated `comp_data` struct to the print function](https://github.com/yankee14/drone_sensor_array/blob/5d1ab9f32cb9b9ec88db85cf0ff885423e0a0a3e/src/drone_sensor_array.c#L265):
```
21.95, 66828.97, 72.69
```
The next loop iteration:
```
RD: 247, 8 0x80 0x0 0x0 0x80 0x0 0x0 0x80 0x0
21.95, 66828.97, 72.69
```
Confirming values:
[image]

Thank you for your help so far!

BoschSensortec, 2017-11-03:
@yankee14 ,

Observing the log for normal mode stream, 
>Then the normal mode stream() line:
>
>RD: 244, 1 0x0
>WR: 242, 1 0x1
>RD: 244, 1 0x0
>WR: 244, 1 0x0
>RD: 244, 1 0x0
>WR: 244, 1 0x54 
**This write didn't work**
>RD: 245, 1 0x0
>WR: 245, 1 0x30
>RD: 244, 1 0x0 
**This read should be 0x54**
>RD: 244, 1 0x0
>WR: 244, 1 0x3
>Temperature, Pressure, Humidity
>RD: 247, 8 0x80 0x0 0x0 0x80 0x0 0x0 0x80 0x0
>21.95, 66828.97, 72.69

It seems that the write is not working as expected. The implementation for the write should look like below, basically write all bytes in a single transaction (refer the table )
```c
int8_t user_i2c_write(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t len)
{
    int8_t rslt = 0; /* Return 0 for Success, non-zero for failure */

    /** Insert this before the write code */
    printf("WR: %d, %d", reg_addr, len);
    for(uint16_t idx = 0; idx < len; idx++)
        printf(" 0x%x", reg_data[idx]);
    printf("\r\n");

    /*
     * The parameter dev_id can be used as a variable to store the I2C address of the device
     */

    /*
     * Data on the bus should be like
     * |------------+---------------------|
     * | I2C action | Data                |
     * |------------+---------------------|
     * | Start      | -                   |
     * | Write      | (reg_addr)          |
     * | Write      | (reg_data[0])       |
     * | Write      | (....)              |
     * | Write      | (reg_data[len - 1]) |
     * | Stop       | -                   |
     * |------------+---------------------|
     */

    uint32_t reg_32[len+1];
    uint32_t reg_32[0] = reg_addr & 0xFF;

    for(uint32_t i = 0; i < len; i++)
        reg_32[i+1] = reg_data[i] & 0xFF;

    rslt = I2C0write(dev_id, reg_32, len+1, 1);
    if(!rslt)
        return -1;

    return 0;
}
```

yankee14, 2017-11-05:
Thank you for catching that error on my part. It did not occur to me that my separate calls to my I2C write function would cause a repeated start condition, which causes the failure, since the writes have to be in sequence. In a "single transaction," as you said.

Thanks for your help!

closed 2017-11-05 by yankee14