boschsensortec/BME280_SensorAPI #32: Cycling values

epvuc, 2017-12-04:
I'm trying this code unchanged on STM32 (stm32f103c8t6), having just added the necessary I2C read and write functions and the delay_ms function.  Using the stream_sensor_data_normal_mode() and stream_sensor_data_forced_mode() examples verbatim from the README in this repo, I see both temperature and pressure readings cycling continuously with a period of about 2.6 seconds (when sampling at 25hz as in the example). 

[image]

[image]

The period of the cycling appears to vary with filter and/or oversampling settings, too, but values never stablilize.  The humidity reading does not cycle, and seems plausible and stable.  The values returned for temperature and pressure are also both within a plausible range and respond to real-world environmental changes.  It happens in both integer and float modes. 

To rule out the stm32 i2c implementation, I repeated the experiment on an ESP32 dev board in the esp-idf environment, implementing esp32-compatible i2c-read/write/delay functions, and am seeing the same cycling.  I haven't found any combinating of oversample, standby, or filter settings that don't cycle.

bme280_init() succeeds and there are no errors returned from any of the functions in bme280.c during operation, fetching calibration data works fine, etc. Saleae dumps of i2c traffic don't show anything out of the ordinary at first glance. 

Any idea of what to look at? I can't even decide if it looks like a software or electrical problem. Power supply seems fine, and anyway stm32 and esp32 boards have different power sources. There are caps to ground close to the Vdd and Vddio pins. There are not cyclic changes in pressure and temperature happening here, at least not at this rapid rate. The sensor does not appear to be measurably light-sensitive. And the swings are very large. Do I have a damaged or defective sensor?  Is this a known failure mode?

Will next try to set up arduino and try one or more of the arduino implementations for comparison. 

```c
void my_print_readings(void)
{
  int8_t rslt = BME280_OK;
  bme280.dev_id = BME280_I2C_ADDR_SEC;
  bme280.intf = BME280_I2C_INTF;
  bme280.read = (void *)user_i2c_read;
  bme280.write = (void *)user_i2c_write;
  bme280.delay_ms = (void *)user_delay_ms;
  rslt = bme280_init(&bme280);
  printf("init result %d\r\n", rslt);
  HAL_Delay(100);
  stream_sensor_data_normal_mode(&bme280);
}
```

```c
int8_t user_i2c_write(uint8_t dev_addr, uint8_t reg_addr, uint8_t *reg_data, uint8_t cnt)
{
  HAL_StatusTypeDef status = HAL_OK;  

  status = HAL_I2C_Mem_Write(&hi2c1, dev_addr<<1, reg_addr, I2C_MEMADD_SIZE_8BIT, reg_data, cnt, 0xffff);
  if (status == HAL_OK)
    return(0);
  else
    return(-1);
}
```

```c
int8_t user_i2c_read(uint8_t dev_addr, uint8_t reg_addr, uint8_t *reg_data, uint8_t cnt)
{
  HAL_StatusTypeDef status = HAL_OK;
  
  status = HAL_I2C_Mem_Read(&hi2c1, dev_addr<<1, reg_addr, I2C_MEMADD_SIZE_8BIT, reg_data, cnt, 0xffff);
  if (status == HAL_OK) 
    return(0);
  else
    return(-1);
}
```

BoschSensortec, 2017-12-04:
Hi @epvuc,

Can you send us the calibration data, the register data read from the sensor (could also be a printf before HAL_I2C_Mem_Write and after HAL_I2C_Mem_Read), and the csv export of the saleae logs?

In addition, which breakout board of the bme280 are you using?

epvuc, 2017-12-04:
Thanks for the rapid response! I'll paste some more complete logs including calibration data and register dumps here, but before wasting more of anyone's time on this I want to verify there's not something electrical to blame. The breakout isn't a commercial one, I'm just using a spare footprint on a generic LGA breakout and some point-to-point wiring for i2c and power to the stm32 dev board. There are local bypass caps but I'm noticing a variation in the amplitude of the cycling when I parallel more with them, which leads me to think that needs to be taken care of before anything else. 

Anyway, log included here. 
[bmelog.txt](https://github.com/BoschSensortec/BME280_driver/files/1526238/bmelog.txt)

BoschSensortec, 2017-12-04:
@epvuc ,

Change in the input voltage during run-time is known to have an effect on the output values. If you have an analog saleae or an oscilloscope, can you check the input voltage variations?