boschsensortec/BME280_SensorAPI #35: In stream_sensor_data_forced_mode the i2ctransfer function fails for burst read

faridabt, 2018-01-24:
In int8_t stream_sensor_data_forced_mode(struct bme280_dev *dev), while burst reading in bme280_get_sensor_data(BME280_ALL, &comp_data, dev); , i2ctransfer stops.  Can you please guide me how to perform i2c burst read??

I tried reading individual registers, but always got default value 0x80000, even after settings and powered on senor. Help me to understand where I am going wrong and how to read actuated values...

Thanks
farida.

BoschSensortec, 2018-01-24:
Hi @faridabt,

Can you share your snippet where you configure the sensor?

dhinojosac, 2018-07-11:
I have the same problem. I always return the same values when reading the data registers (0xF7-0xFE) after configuring the module. 
`       configure_i2c_master(&i2c_master_instance);
	

	int8_t rslt = BME280_OK;

	my_dev.dev_id		= BME280_I2C_ADDR_SEC;
	my_dev.intf		= BME280_I2C_INTF;
	my_dev.read		= user_i2c_read;
	my_dev.write		= user_i2c_write;
	my_dev.m_delay_ms	= user_delay_ms;

	rslt = bme280_init(&my_dev);
	if (rslt==BME280_OK)
	{
		dbg_prints("BME280 Init success");
	}

	stream_sensor_data_normal_mode(&my_dev);`

[image]

BoschSensortec, 2018-07-12:
Hi @dhinojosac,

- Does the user_delay_ms wait for the required amount of time? 
- How long do you wait before each read?

dhinojosac, 2018-07-12:
I had a problem writing the registers!.
I share my functions!
```
int8_t user_i2c_write(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t len)
{
    int8_t rslt = 0; 
	dohc_i2c_master_write_byte(&i2c_master_instance,dev_id,reg_addr,*reg_data);
    return rslt;
}
```

```
enum i2c_master_result dohc_i2c_master_write_byte(struct i2c_master_module  * module, uint8_t dev_id, uint8_t addr, uint8_t data)
{
	uint16_t timeout = 0;
	uint8_t buffer[10];
	// Clear write buffer
	memset(buffer, 0, 10);

	// Set first byte of write buffer to sensor reg address
	buffer[0] = addr;

	// Set second byte of write buffer to data
	buffer[1] = data;

	/* Init i2c packet. */
	struct i2c_master_packet packet =
	{
		.address = dev_id,
		.data_length = 2,
		.data = buffer,
		.ten_bit_address = false,
		.high_speed = false,
		.hs_master_code = 0x0,
	};

	// Write data to sensor
	while (i2c_master_write_packet_wait(module, &packet) != STATUS_OK)
	{
		// Increment timeout counter and check if timed out.
		if (timeout++ == 1000)
		{
			dbg_prints("timeout_read");
			break;
		}
	}

	return I2C_MAST_OK;
}
```

BoschSensortec, 2018-07-16:
Hi @dhinojosac ,

You seem to be writing only one register in the user_i2c_write. You should use the whole array or `len` bytes pointed by `reg_data`.