boschsensortec/BME280_SensorAPI #36: Documentation is unclear on units/format of output for pressure/humidity readings

mhord, 2018-01-31:
When calling `bme280_get_sensor_data()` and looking at the data returned in the `bme280_data` struct, it's unclear what the units/format of the returned data are. For instance, temp is returned as degrees C * 100.

Can you clarify for me?

BoschSensortec, 2018-02-01:
Hi @mhord,

Depending on whether the C flag, -DBME280_FLOAT_ENABLE is passed, or if the [related line](https://github.com/BoschSensortec/BME280_driver/blob/01ec205320b56a42f226e162595d1efc8a797cf6/bme280_defs.h#L105) is  not commented in the header file,

In case of the flag enabled,
The outputs are in ```double``` and the units are 
 - °C for temperature
 - % relative humidity
 - Pascals for pressure

In case of the flag not enabled, then it is
 - ```int32_t``` for temperature with the units 100*°C 
 - ```uint32_t``` for humidity with the units 1000*% relative humidity
 - ```uint32_t``` for pressure
   - with the units 100 * Pascals if [this define](https://github.com/BoschSensortec/BME280_driver/blob/01ec205320b56a42f226e162595d1efc8a797cf6/bme280_defs.h#L110) is enabled, which it is by default
   - or if that line is commented in the header, it is in Pascals.

Hope this clarifies it and doesn't confuse you further. :)

mhord, 2018-02-01:
Thank you for the clarification and the prompt response!

closed 2018-02-01 by mhord

BoschSensortec, 2018-02-26:
@mhord,

There was a mistake regarding the units for humidity as another use pointed out. Refer the readme's chapter for the units.
https://github.com/BoschSensortec/BME280_driver/tree/bme280_v3.3.4#sensor-data-units