boschsensortec/BME280_SensorAPI #37: Sensor Readout - Registers only show default values

rwon88, 2018-02-06:
Dear all,

I'm trying to read out the BME280 over the I2C Interface.
I configure the sensor upfront as follows:
RESET = 0xB6
CONFIG_REGISTER = 0xE5
CTRL_MEAS = 0x93

After that I'm trying to readout the sensor, but I only get the default values back (According to datasheet page 25.
PRESSURE_LSB = 0x00
PRESSURE_MSB = 0x80
HUMIDITY_LSB = 0x00
HUMIDITY_MSB = 0x80
TEMPERATURE_LSB = 0x00
TEMPERATURE_MSB = 0x80

I tried burst read and seperate read, both Show the same result. It seems to me that the sensor is actually not sampling.
I use my own Code. 
Can anybody give me some hint, what I'm doing wrong?

Thanks in advance and kind regards

Ron

BoschSensortec, 2018-02-07:
@rwon88,

Can you verify that you have the expected values written to the ```CONFIG_REGISTER``` and ```CTRL_MEAS``` ?

rwon88, 2018-02-09:
Reading the Registers CONGIG_REGISTER, CTRL_MEAS and CTRL_HUM shows 0x00. So they seem not be written. The sequence of writing CTRL_HUM first and CTRL_MEAS after that has been applied.
I also tried to put some delay between the writing of CTRL_HUM and CTRL_MEAS.
What am I'm doing wrong?

Kind Regards

Ron

BoschSensortec, 2018-02-09:
@rwon88,

You will have to check the bus communication using a logic analyzer like that from [Saleae](https://www.saleae.com/) to see if you receive the right acknowledgements. Usually it could be an issue of the low level driver implementation, the write function.

dhinojosac, 2018-07-11:
I have the same problem. I always return the same values when reading the data registers (0xF7-0xFE) after configuring the module. 

`	configure_i2c_master(&i2c_master_instance);	

	int8_t rslt = BME280_OK;

	my_dev.dev_id		= BME280_I2C_ADDR_SEC;
	my_dev.intf		= BME280_I2C_INTF;
	my_dev.read		= user_i2c_read;
	my_dev.write		= user_i2c_write;
	my_dev.m_delay_ms	= user_delay_ms;

	rslt = bme280_init(&my_dev);
	if (rslt==BME280_OK)
	{
		dbg_prints("BME280 Init success");
	}
	
	stream_sensor_data_normal_mode(&my_dev);`

[image]