boschsensortec/BME280_SensorAPI #38: compensation data for temperature is zero

vkiselyov, 2018-02-08:
Hi.
I have some problem with temperature measurement.
When I read result om measurement from sensor i have value for all parameters(temp, pres, hum).
But after compensation function temperature became 0.
Data at 0xF7 - 0xF9 are not zero valued.
I check comp_data in reg 0x88 - 0x8D they are 0.
function compensate_temperature use this zero valued data and temperature became 0;

What I have to do for get temperature value?

BoschSensortec, 2018-02-08:
Hi @vkiselyov,

Which version of the driver are you using? Can you share your configuration snippet?

vkiselyov, 2018-02-08:
Hi. Thank for you attention.
 @file	bme280.h
 * @date	22 Nov 2017
 * @version	3.3.2
 * @brief

dev.settings.osr_h = BME280_OVERSAMPLING_1X;
	dev.settings.osr_p = BME280_OVERSAMPLING_16X;
	dev.settings.osr_t = BME280_OVERSAMPLING_2X;
	dev.settings.filter = BME280_FILTER_COEFF_16;
	dev.settings.standby_time = BME280_STANDBY_TIME_1000_MS;

BoschSensortec, 2018-02-08:
@vkiselyov,

Are the following settings same or similar?

```c
	settings_sel = BME280_OSR_PRESS_SEL;
	settings_sel |= BME280_OSR_TEMP_SEL;
	settings_sel |= BME280_OSR_HUM_SEL;
	settings_sel |= BME280_STANDBY_SEL;
	settings_sel |= BME280_FILTER_SEL;
	rslt = bme280_set_sensor_settings(settings_sel, dev);
	rslt = bme280_set_sensor_mode(BME280_NORMAL_MODE, dev);
```

vkiselyov, 2018-02-08:
settings_sel = BME280_OSR_PRESS_SEL;
	settings_sel |= BME280_OSR_TEMP_SEL;
	settings_sel |= BME280_OSR_HUM_SEL;
	settings_sel |= BME280_STANDBY_SEL;
	settings_sel |= BME280_FILTER_SEL;

rslt = bme280_init(&dev);
rslt = bme280_set_sensor_settings(settings_sel, &dev);
rslt = bme280_set_sensor_mode(BME280_NORMAL_MODE, &dev);

BoschSensortec, 2018-02-08:
@vkiselyov ,

Can you make a read/write log? Refer https://github.com/BoschSensortec/BME680_driver/issues/14 for an idea on how to do it.

vkiselyov, 2018-02-08:
RD: 208, 1 0x60
WR: 224, 1 0xb6
RD: 136, 26 0x0 0x0 0x0 0x0 0x0 0x0 0xfd 0x8f 0x3b 0xd7 0xd0 0xb 0x55 0x1e 0x4a 
0xff 0xf9 0xff 0xc 0x30 0x20 0xd1 0x88 0x13 0x0 0x4b

RD: 225, 7 0x45 0x1 0x0 0x1a 0x27 0x3 0x1e
RD: 244, 1 0x0
WR: 242, 1 0x1
RD: 244, 1 0x0
WR: 244, 1 0x0
RD: 244, 1 0x0
WR: 244, 1 0x54
RD: 245, 1 0x0
WR: 245, 1 0xb0
RD: 244, 1 0x54
RD: 244, 1 0x54
WR: 244, 1 0x57
RD: 247, 8 0x50 0x45 0x80 0x82 0x7 0x80 0x80 0x17
 comp_data D - 64, 1233125376, 1063884786
RD: 247, 8 0x50 0x45 0x40 0x82 0x5 0xe0 0x7f 0xf4
RD: 247, 8 0x50 0x45 0x50 0x82 0x4 0x90 0x80 0x9
RD: 247, 8 0x50 0x45 0xa0 0x82 0x5 0x50 0x80 0x1
RD: 247, 8 0x50 0x45 0xe0 0x82 0x5 0xf0 0x7f 0xb7
RD: 247, 8 0x50 0x45 0xe0 0x82 0x5 0x40 0x7f 0x7b

BoschSensortec, 2018-02-08:
@vkiselyov 

Which compiler are you using and for what target?

vkiselyov, 2018-02-08:
gcc 6.3.0 STM32

vkiselyov, 2018-02-12:
@BoschSensortec 
Hi.
Do you have any idea?

BoschSensortec, 2018-02-13:
Hi @vkiselyov,

No idea so far, can you try to enable the floating point compensation by un-commenting this line https://github.com/BoschSensortec/BME280_driver/blob/18260d616b8c03fc1c4e1d68bed578c4605d40b4/bme280_defs.h#L105 

You can refer to https://github.com/BoschSensortec/BME280_driver/issues/36 for the units.

If this works, then it might have something to do with the shifting operations.

vkiselyov, 2018-02-14:
[image]

factory settings temperature calibration for device is 0

static int32_t compensate_temperature(const struct bme280_uncomp_data *uncomp_data,
						struct bme280_calib_data *calib_data)
{
	int32_t var1;
	int32_t var2;
	int32_t temperature;
	int32_t temperature_min = -4000;
	int32_t temperature_max = 8500;

	var1 = (int32_t)((uncomp_data->temperature / 8) - ((int32_t)calib_data->dig_T1 * 2));
	var1 = (var1 * ((int32_t)calib_data->dig_T2)) / 2048; 
	var2 = (int32_t)((uncomp_data->temperature / 16) - ((int32_t)calib_data->dig_T1));
	var2 = (((var2 * var2) / 4096) * ((int32_t)calib_data->dig_T3)) / 16384; 
	calib_data->t_fine = var1 + var2; **t_fine =0 becase 0+0 = 0**
	temperature = (calib_data->t_fine * 5 + 128) / 256;

	if (temperature < temperature_min)
		temperature = temperature_min;
	else if (temperature > temperature_max)
		temperature = temperature_max;

	return temperature;
}

**/VAR1 = 0 because dig_T2 = 0/**
**var 2 = 0 because dig_T3=0**
 **RESULT = 0**

vkiselyov, 2018-02-14:
When I use #define BME280_FLOAT_ENABLE
i receive result: 
comp_data HEX 403b46f3, 0, 0

BoschSensortec, 2018-02-14:
Hi @vkiselyov,

Almost all values from the factory calibration should be non-zero. Can you send us information on the marking on that part? Refer Chapter 7.5 of the [datasheet](https://ae-bst.resource.bosch.com/media/_tech/media/datasheets/BST-BME680-DS001-00.pdf).

vkiselyov, 2018-02-14:
663 UP

BoschSensortec, 2018-02-14:
@vkiselyov,
Seems to be an issue with the sensor. Do you have another sensor to test with?

BoschSensortec, 2018-02-14:
@vkiselyov, 

Also, can you send us the register context 0x83 to 0x86.

vkiselyov, 2018-02-14:
0x83 - 7e
0x84 - 8c
0x85 - 1e
0x86 - f1

HanVertegaal, 2018-02-27:
Hi,
Any news on this? I have exactly the same problem. The temperature calibration values are all zero. Pressure and Humidity are working fine.

vkiselyov, 2018-02-27:
Hi. I Set new sensor and have non-zero values.
But now temperature is always 8500.

HanVertegaal, 2018-02-27:
Thanks for the update.
I have the parts for a new module, but I guess I will need to review the soldering instructions before assembling it. Still, it seems odd to me that only the temperature is affected, and not the humidity or pressure.

HanVertegaal, 2018-02-27:
Also, I do seem to get valid uncompensated temperature data, so the sensor itself works. It's just the calibration data.

vkiselyov, 2018-02-27:
probably it defective batch

BoschSensortec, 2018-02-28:
@vkiselyov & @HanVertegaal ,
The observations show that is a defective batch. Where were these sensor's purchased from?

HanVertegaal, 2018-02-28:
Mine were purchased from Mouser Electronics Nederland (Eindhoven).

HanVertegaal, 2018-03-26:
Extending the delay in **_bme280_soft_reset_** to 20 ms solved the calibration values readout problem.

The device is now working as expected.

BoschSensortec, 2018-03-26:
Hi @HanVertegaal,

Thank you for the insight. Can you check if your linked `delay_ms` function is accurate to the allotted time?

HanVertegaal, 2018-03-26:
Indeed it is not. It's resolution is 10 ms, which explains why the 2 ms delay in `bme280_soft_reset` did not have the desired effect. but the 20 ms did.

klaus-liebler, 2018-11-08:
@HanVertegaal , thank you for your contribution in this discussion! I am working with the esp-idf ("latest") for Espressif ESP32 and have had the same problem. After having updated the soft_reset_timeout to 20ms, everything works perfectly:-)