boschsensortec/BME280_SensorAPI #40: Possibly improper value returned by integer version of compensate_humidity function

m-xor, 2018-02-13:
According to explanation given in #36 function _compensate_humidity_ returns  1000 * % relative humidity. It's inconsistent with data sheet. Comparison to function _bme280_compensate_H_int32_ in data sheet reveals that both functions are practically the same. But according to commentary,  function _bme280_compensate_H_int32_: "Returns humidity in %RH as unsigned 32 bit integer in Q22.10 format (22 integer and 10 fractional bits). Output value of “47445” represents 47445/1024 = 46.333 %RH".  Also pre final test of upper limit in function _compensate_humidity_ tests value of 419430400/4096=102400, not 100000. So, it appears that function returns rather 1000/1024 * % relative humidity.

BoschSensortec, 2018-02-26:
@m-xor,

Refer to the latest release for the corrections.

m-xor, 2018-02-26:
Everything is clear now. Thank you

closed 2018-02-26 by m-xor