boschsensortec/BME280_SensorAPI #43: Bme280  Bmp080

comarius, 2018-03-15:
I have 2 types of sensors, BME280 sparkfun and bmp 080
I am using same i2c lib for both.
The code for both is form the samples provided.

The bmp080 shows data, while bme280 every time the program is restarted shows different data.
They sit side by side:

Bmp080  T: 227       P: 97619
BMe280  T: -2322   P: 110000    H: 100000  
Any hints ?

BoschSensortec, 2018-03-15:
Hi @comarius,

Which software libraries are you using for both? Can you provide the links here along with the links to the boards?

Thank you.

comarius, 2018-03-16:
Hi,
Bmp080 from amazon: https://www.amazon.ca/Puuli-Atmospheric-Pressure-Module-Altimeter/dp/B018VN2EZQ/ref=sr_1_5?ie=UTF8&qid=1521159473&sr=8-5&keywords=bmp085
which works OK.

Bme280 https://www.sparkfun.com/products/13676.
Which after every power on, gave me different bogus results.

I thought is the sensor but works ok whit this code:
https://www.blackmagicboxes.com/?p=608

Both sensors are using same I2C read/write and delay callbacks.
I am using this library:

https://github.com/yigityuce/BlackLib/tree/master/v3_0/BlackI2C
and the callbacks are using writeBlock/readBlock from this i2c.

The code to read Bme280 is this one:

https://github.com/BoschSensortec/BME280_driver  down the page at section:
int8_t stream_sensor_data_normal_mode(struct bme280_dev *dev){....}

My init is:

        _bme280.intf = BME280_I2C_INTF;
        _bme280.delay_ms = &delay;
        _bme280.read = &Bme_280::_iread;
        _bme280.write = &Bme_280::_iwrite;
          ::bme280_init(&_bme280); OK returns BME_OK

Bme_280::_iread & Bme_280::_iwrite  as I said calls the ic2lib blockR/W() finally.

Both Bme080 Bmp280 have same structure.

BoschSensortec, 2018-03-16:
Hi @comarius,

Can you direct us to the ```Bme_280::_iread``` & ```Bme_280::_iwrite``` library? The sensor API expects them to perform the writes in a specific manner. Quite often, the write function is incorrect.

comarius, 2018-03-17:
I really appreciate the prompt response.
They just wrappers from c++ class to adding a this pointer, but they pass all parameters here

Here is the call chain from Bme280 Class - > Base class and  -> i2c class. Is the same for write.
(I add to dev structure 'this', as well to read/write functors, so I can C++-it...)

 
     int8_t Bme_280::_iread(uint8_t device_addr,
                                 uint8_t register_addr,
                                 uint8_t *register_data,
                                 uint16_t read_length, void* pthis)
        {
            if(((Bme_280*)pthis)->_ok){
                return ((Bme_280*)pthis)->busread(register_data, read_length, register_addr);
            }
            return 0;
       }
    // base class

    bool    WireBus::busread(uint8_t* buff, int len, int anyreg)
    {
        return _pi2c->readBlock(anyreg, buff, len);
    }

Where readBlock  / writeBlock are in this library.
https://github.com/yigityuce/BlackLib/blob/master/v3_0/BlackI2C/BlackI2C.cpp

to writeBlock / readBlock

BoschSensortec, 2018-03-19:
@comarius,

We are not sure about the i2c smbus implementation, but you can refer to an alternate implementation here in this thread [here](https://github.com/BoschSensortec/BHy1_driver_and_MCU_solution/issues/3#issuecomment-371806056).

comarius, 2018-03-20:
Thank you. Ill let you know soon.

closed 2018-03-26 by comarius