boschsensortec/BME280_SensorAPI #45: Reading two BME280 sensors on the same bus

Stormhand, 2018-03-29:
Hello,

Im trying to use this driver to read two identical sensors on the same I2C bus using both addresses 0x76 and 0x77 in forced mode. 
platform: esp32 esp-idf xtensa

**bme280_init** initializes the static struct bme280_t *p_bme280. The function notes state:

>  *	@note While changing the parameter of the bme280_t
>   *	@note consider the following point:
>  *	Changing the reference value of the parameter
>  *	will changes the local copy or local reference
>  *	make sure your changes will not
>  *	affect the reference value of the parameter
> *	(Better case don't change the reference value of the parameter)

Is there any way to initialize two bme280_t structs to avoid calling bme280_init on each read?

Regards

BoschSensortec, 2018-03-29:
Hi @Stormhand,

The latest version of the API v3.3.4 support multi-instances. The releases can be found [here](https://github.com/BoschSensortec/BME280_driver/releases).

Stormhand, 2018-03-29:
Thank you very much for the fast response! I will test it today and close the issue.

Regards!

closed 2018-03-30 by Stormhand