boschsensortec/BME280_SensorAPI #47: Pressure reading is wrong

TonyTheLion, 2018-04-11:
Hi,

I am using your library to calculate the pressure, but its always wrong.  
The other values produced by the device are correct.
I have set my device to the recommended settings per some of your examples.

I am not seeing why my pressure readings would be off by so much. 

Perhaps you can help?

I'm using Debian on an ARM BeagleBone chip.

Here's the code:

```
#include <stdio.h>
#include <fcntl.h>
#include <sys/ioctl.h>
#include <linux/i2c.h>
#include <linux/i2c-dev.h>
#include <stdio.h>
#include <unistd.h>
#include "read_data.h" /* I2C functionality */
#include "bme280_.h" /* Constants & read & write I2C wrappers */
#include "bme280.h" /* Bosh Sensortec driver header */
#include "bme280_defs.h" /* Bosh Sensortec driver header */

int main()
{
	/* Using Bosh driver and my own code */
	struct bme280_dev device;
	int8_t rslt = BME280_OK;

	/* Set device settings */
	device.dev_id = BME280_I2C_ADDRESS; // 0x77
	device.intf = BME280_I2C_INTF;
	device.read = read_reg_bosh;
	device.write = write_reg_bosh;
	device.delay_ms = delay_bosh;

	/* Desired device settings that are needed to be set */
	device.settings.osr_p = BME280_OVERSAMPLING_16X;
	device.settings.osr_t = BME280_OVERSAMPLING_2X;
	device.settings.osr_h = BME280_OVERSAMPLING_1X;
	device.settings.filter = BME280_FILTER_COEFF_16;
	device.settings.standby_time = BME280_STANDBY_TIME_62_5_MS;

	rslt = bme280_init(&device);
	if (rslt != BME280_OK)
	{
		printf("Error occurred!\r\n");
		printf("result: %d\r\n", rslt);
	}

	/* Set sensor settings */
	uint8_t sensor_settings = BME280_OSR_PRESS_SEL; /* Set which settings to set.  Set to 'indoor navigation' as per README.md. */
	sensor_settings |= BME280_OSR_TEMP_SEL;
	sensor_settings |= BME280_OSR_HUM_SEL;
	sensor_settings |= BME280_STANDBY_SEL;
	sensor_settings |= BME280_FILTER_SEL;
	rslt = bme280_set_sensor_settings(sensor_settings, &device);
	if (rslt != BME280_OK)
	{
		printf("Error writing sensor settings\r\n");
		return -1;
	}

	/* Sensor settings are written to device->settings */
	rslt = bme280_get_sensor_settings(&device);
	if (rslt != BME280_OK)
	{
		printf("Error reading sensor settings\r\n");
		return -1;
	}

	/* Print settings */
	printf("Pressure oversampling: %x\r\n", device.settings.osr_p);
	printf("Temperature oversampling: %x\r\n", device.settings.osr_t);
	printf("Humidity oversampling: %x\r\n", device.settings.osr_h);
	printf("Filter: %x\r\n", device.settings.filter);
	printf("Standby Time: %x\r\n", device.settings.standby_time);

	/* Turn on device in normal mode */
	rslt = bme280_set_sensor_mode(BME280_NORMAL_MODE, &device);
	if (rslt != BME280_OK)
	{
		printf("Error during power on\r\n");
		printf("result %d\r\n", rslt);
	}

	uint8_t buff = 0;
	rslt = bme280_get_sensor_mode(&buff, &device);
	if (rslt != BME280_OK)
 	{
		printf("Error during read reg\r\n");
		printf("result %d\r\n", rslt);
 	}

	if (buff == BME280_NORMAL_MODE)
 	{
		printf("Device Powered On\r\n");
	}

	/* Delay while sensor completes measurement. */
	device.delay_ms(70);

	/* Read sensor data & compensate data from sensor */
	uint8_t sensor_comp = BME280_ALL; /* Which values to compensate */
	struct bme280_data comp_data;
	rslt = bme280_get_sensor_data(sensor_comp, &comp_data, &device);
	if (rslt != BME280_OK)
	{
		printf("Error reading sensor data\r\n");
		return -1;
	}

	/* Output data */
	printf("Pressure: %f\r\n", comp_data.pressure);
	printf("Temperature: %f\r\n", comp_data.temperature);
	printf("Humidity: %f\r\n", comp_data.humidity);
	return 0;
}
```
Here's the output:
```
Pressure oversampling: 5
Temperature oversampling: 2
Humidity oversampling: 1
Filter: 4
Standby Time: 1
Device Powered On
Pressure: 67893.217459
Temperature: 22.481205
Humidity: 58.589021
```

BoschSensortec, 2018-04-11:
Hi @TonyTheLion,

Is this the case for all the readings? Can you also print out the calibration parameters from bme280_dev struct? You can find it under `device.calib_data.dig_`.

TonyTheLion, 2018-04-11:
Hi,

Yes all readings of pressure are pretty much the same value you see.

Calibration Parameters:

```
dig_T1: 28333
dig_T2: 26573
dig_T3: 50
dig_P1: 37362
dig_P2: -10666
dig_P3: 3024
dig_P4: 7253
dig_P5: 50
dig_P6: -7
dig_P7: 9900
dig_P8: -10230
dig_P9: 4285
dig_H1: 75
dig_H2: 352
dig_H3: 0
dig_H4: 343
dig_H5: 0
dig_H6: 30
t_fine: 115103
```

TonyTheLion, 2018-04-12:
In fact, looking at this again, it seems that all three values (Pressure, Humidity, Temperature) are always the same.

Today the temperature in the room where this device lives (per my thermostat) is 21 C, the BME device still shows it to be 22.481.

BoschSensortec, 2018-04-13:
@TonyTheLion,

Which breakout board of the BME280 are you using?

TonyTheLion, 2018-04-13:
@BoschSensortec,

https://www.adafruit.com/product/2652

TonyTheLion, 2018-04-20:
Hi @BoschSensortec,

Any more updates on this issue?

BoschSensortec, 2018-04-24:
Hi @TonyTheLion,

We haven't been able to duplicate the issue yet.

In the meanwhile, can you send us your build options?

Thank you.

TonyTheLion, 2018-04-26:
Hi,

I am using `meson` to build my little test program, so I'm going to just give you the commands that `ninja -t commands` gives me.  Hope that is helpful

Currently only doing a debug build.

Compiler flags:
`cc  -Itest/bme280_test@exe -Itest -I../test -Iinclude -I../include -I. -I../ -Icerts -I../certs -I/usr/local//include -fdiagnostics-color=always -pipe -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -std=gnu11 -O0 -g -MMD -MQ 'test/bme280_test@exe/bme280_test.c.o' -MF 'test/bme280_test@exe/bme280_test.c.o.d' -o 'test/bme280_test@exe/bme280_test.c.o' -c ../test/bme280_test.c`

Linker flags:
`:cc  -o test/bme280_test 'test/bme280_test@exe/.._src_read_data.c.o' 'test/bme280_test@exe/.._src_bme280.c.o' 'test/bme280_test@exe/.._src_bme280_.c.o' 'test/bme280_test@exe/bme280_test.c.o' -Wl,--no-undefined -Wl,--as-needed`

I have recently gotten a new physical BME device, but it still gives me the wrong pressure.  So it seems likely that either I'm doing something wrong, or there's an error in the software somewhere.

BoschSensortec, 2018-05-02:
Hi @TonyTheLion,

- If the readings are not changing between calls,
  - The wait period is too little between read outs
  - The settings haven't been set correctly to the sensor
- If the readings are changing between calls but the values are not accurate,
  - It seems to be an issue with how the compiler interprets the shifts and typecasts in the compensation equations. You will have to compare results between integer and floating point to see if they match.

ThomasLeMezo, 2018-05-16:
Hi,
I have the exact same issue with a Raspberry Pi 3. The problem seems to be link with the use of these two functions:
```
rslt = bme280_set_sensor_mode(BME280_NORMAL_MODE, &dev);
rslt = bme280_set_sensor_settings(settings_sel, &dev);
```
If I first use a python code provided here [to](https://bitbucket.org/MattHawkinsUK/rpispy-misc/raw/master/python/bme280.py) to configure the sensor and the run the c program without the two lines it works perfectly.
It seems that when configuring the sensor, it is not set in measuring mode as values stay constant with the two above functions.

ThomasLeMezo, 2018-05-17:
Hi @BoschSensortec,
I tried your code but I have still the same issue.
Any other ideas?

Thanks

BoschSensortec, 2018-05-17:
@ThomasLeMezo,

What your build options?

ThomasLeMezo, 2018-05-19:
@BoschSensortec 
I don't think it is an issue with the compiler build option. The sensor seems to stay in SLEEP_MODE after using `bme280_set_sensor_settings` or `bme280_set_sensor_mode`.
I'm compiling the code for a ROS node so I use the default option provided (gcc, CMake).

ThomasLeMezo, 2018-05-23:
@TonyTheLion,
Thanks for the feedback, it seems it solves the issue!