boschsensortec/BME280_SensorAPI #49: BME280.c Compile Problem

srf461, 2018-05-07:
Good morning all.  This is my first post here.  I am a C rookie and am trying to incorporate a BME280 sensor into a project of mine that is using a PIC18F27K42 MCU.  My IDE is MPLAB X IDE.  My compiler is XC8.  I wrote and successfully built my main() function.  Then I copied bme280.c, bme280.h, and bme280_defs.h from github:    

https://github.com/BoschSensortec/BME280_driver

I incorporated them into my project.  I did NOT call the bme280.c from my main() function, I simply incorporated the files and tried to get it to compile.

It did not.  It gave me a bunch of warnings and two errors.  The first error was as follows:
Compiler Error:   BME280.c:594: error: (188) constant expression required
bme280.c line 594:  	uint8_t reg_data[BME280_P_T_H_DATA_LEN] = {0};

I tried taking a systematic approach to it by looking at each element of the line individually:
I found that "uint8_t" that is in the line is not defined in bme280.c or in the other two files.  I went out on the internet and found an "inttypes.h" file that defines it as "typedef unsigned char uint8_t;.  I incorporated that file into my project and did a #include "inttypes" in bme280.c.  Then I built it again.  I got the same error with a different line number of course because of the addition of the #include statement.

Then I looked at what I assume is a constant: "BME280_P_T_H_DATA_LEN".  I found it defined in bme280_defs.h in the following statement: 

#define BME280_P_T_H_DATA_LEN				UINT8_C(8)

And finally, what appears to be the array I assume is locally defined on that line.

Finally, it looks as if others have used this code successfully.  So what am I doing wrong?  

Thanks in advance,
Ray Russell

srf461, 2018-05-07:
And here is the second error.  It has the same look and feel as the first one:

This is the compiler error message:   BME280.c:1153: error: (188) constant expression required

And this is the line number 1153:  uint8_t calib_data[BME280_TEMP_PRESS_CALIB_DATA_LEN] = {0};

The compiler manual's entry regarding error 188 did not provide me much help.  However, as I said at the top I'm a rookie at this...

BoschSensortec, 2018-05-08:
Hi @srf461,

`uint8_t` and `UINT8_C` are standard C99 types which should be found in the standard library headers, `stdint.h` and `stddef.h`. In the BME280 API, it is included in the `bme280_defs.h` file as seen [here](https://github.com/BoschSensortec/BME280_driver/blob/958cef65daa46123bdc3a8da9fb2a1b0abffc653/bme280_defs.h#L64). 

Make sure you are using the C99 standard or higher.

srf461, 2018-05-08:
Good morning and thank you for your very quick reply.
I don't know how I missed the uint8_t definition.  Thank you for taking the time to point me to it.

I looked at my compiler documentation and found that it complies with C89 but "some features of the later standard, C99 are also supported". 

I went back and deleted my versions of the BME280.c, .h, and BME280_defs.h files and replaced them with unmodified versions as created by Bosch.  After recompiling I got a large number of errors, the first of which was essentially the same error that I got before: 

Compiler error: BME280.c:592: error: (188) constant expression required
Source line:   	uint8_t reg_data[BME280_P_T_H_DATA_LEN] = {0};

I still don't know what is wrong with that line although the uint8_t element is clearly not the issue.  

Other compile errors: 
BME280.c:1019: error: (192) undefined identifier "int64_t"
Associated source file line: # 1019:  	int64_t var1;
My guess is that this is another instance where my compiler is not compliant with C99?  I found that the version of "inttypes.h" that I had found on the internet has the "int64_t" defined in it.  So i #include'd it and recompiled.  All of the compile errors disappeared except for my original two which are:

1) Compiler error statement: "BME280.c:593: error: (188) constant expression required"
Associated source file line: #593:  	"uint8_t reg_data[BME280_P_T_H_DATA_LEN] = {0};"

2) Compiler error statement: "BME280.c:1152: error: (188) constant expression required"
Associated source file line #1152:  "uint8_t calib_data[BME280_TEMP_PRESS_CALIB_DATA_LEN] = {0};"

Now since I copied the code directly from the github (Bosch) files and my only modification was to #include  "inttypes.h" to fix the "int64_t" issue I must conclude that these two errors are stemming from my use of the MCU manufacturer's compiler.   I looked but did not find any other compilers for my chosen MCU chip.  

I have continued looking into the problem and noticed that in bme280_defs.h there is a statement: 
     "BME280_P_T_H_DATA_LEN       UINT8_C(8)"

And in stdint.h there is a statement: 
     "#define UINT8_C(c) __UINT8_C(c)"

However, I don't see where __UINT8_C(c) is defined.  I think this could be a macro of some sort???  Could this be the missing piece?
      "

BoschSensortec, 2018-05-08:
Hi @srf461,

Could you try to see which of the following solves the issue?
- Remove the casts
```c
#define BME280_TEMP_PRESS_CALIB_DATA_LEN  (26)
#define BME280_HUMIDITY_CALIB_DATA_LEN    (7)
#define BME280_P_T_H_DATA_LEN             (8)
```
- or, remove the initializer
```c
uint8_t reg_data[BME280_P_T_H_DATA_LEN];
```

srf461, 2018-05-08:
Hi,
When I removed the casts i got the following error:
BME280.c:592: error: (192) undefined identifier "BME280_P_T_H_DATA_LEN"
BME280.c:592: error: (188) constant expression required
Line 592 is "	uint8_t reg_data[BME280_P_T_H_DATA_LEN] = {0};"

When I removed line 592 AND with the casts removed I got the following two errors on line 600:
BME280.c:600: error: (192) undefined identifier "reg_data"
BME280.c:600: error: (192) undefined identifier "BME280_P_T_H_DATA_LEN"

Line 600 is: "		rslt = bme280_get_regs(BME280_DATA_ADDR, reg_data, BME280_P_T_H_DATA_LEN, dev);"

srf461, 2018-05-08:
Oh wait, I will put the casts back in with the identifier out...I will report back in a moment

srf461, 2018-05-08:
With the three casts removed and the initializer still in place i get the following errors and warnings:

BME280.c:600: error: (192) undefined identifier "reg_data"
BME280.c:600: warning: (357) illegal conversion of integer to pointer
BME280.c:600: error: (192) undefined identifier "BME280_P_T_H_DATA_LEN"
BME280.c:600: warning: (373) implicit signed to unsigned conversion

Line 600 is: 		rslt = bme280_get_regs(BME280_DATA_ADDR, reg_data, BME280_P_T_H_DATA_LEN, dev);

With the casts in place and the initializer commented out i get the following errors and warnings:

BME280.c:600: warning: (373) implicit signed to unsigned conversion
BME280.c:600: error: (192) undefined identifier "reg_data"
BME280.c:600: warning: (357) illegal conversion of integer to pointer
BME280.c:600: warning: (373) implicit signed to unsigned conversion

Line 600 is: 		rslt = bme280_get_regs(BME280_DATA_ADDR, reg_data, BME280_P_T_H_DATA_LEN, dev);

srf461, 2018-05-08:
Update:
I may have misunderstood what you meant in your last post:  I commented out the three casts.  That did not work.  But what I have JUST done is to replace:
"UINT8_C(26)" with "26"
"UINT8_C(7)" with "7"
"UINT8_C(8)" with "8".

The files compiled successfully.  Is that what you were suggesting?  If so, then that worked.

There are still lots of warnings but if I can get code to load into my MCU then I believe I can troubleshoot it and make it work.

Please respond back and let me know if what I have done matches up with what you intended.

Thank you very much,
Ray Russell
Houston, TX

BoschSensortec, 2018-05-09:
Hi @srf461,

Yes, your last post implements option 1 as intended.  We will in the meantime look into why some compilers do not support these macros.

srf461, 2018-05-09:
Thank you.  I will be interested in hearing what you find out.  As I mentioned earlier my particulars are:
 MCU, IDE, and compiler supplier:  Microchip.com
MCU:  PIC18F27K42
IDE:  MPLAB X IIDE. v4.15.,(free)
Compiler:XC8 for 8-devices (also free)

They also have compilers for 16 and 32-bit devices.. I assume they are free as well.

Thank you again for all your help.
Ray R

srf461, 2018-05-13:
Well, after struggling with this for the past couple of days I've reached the conclusion that I clearly don't know what I'm doing.  To recap:

I added bme280.c, bme280.h, and bme280_defs.h to my project file that contains a small top_level.c routine that holds main().  The files compiled successfully.

Next, I inserted one of the functions from bme280.c into my main() routine, with the intent being attempt to execute it from inside main().  However, the program would not compile without errors.  '

I feel the need at this point to remind the reader that I am a C language neophyte.  

Details:  
The particular file that I picked from bme280.c was "static int8_t get_calib_data(struct bme280_dev *dev", because it seemed less complex than the others.  

It was also used inside the function "int8_t bme280_init(struct bme280_dev *dev)", so I felt like I could use that as kind of a template to guide me.

The particular instance that was used inside "int8_t bme280_init(struct bme280_dev *dev)" was:
"rslt = get_calib_data(dev);".  So, as a test I pasted that into my main() routine and declared "rslt" as an "int8"t" prior to where the function was called.  But that didn't work.  

I think I know that "*dev" is a pointer but I didn't see anything done to it inside the function from where I copied it so I didn't try to declare it.  Maybe that was my error, I don't know.

I would appreciate anything that might point me in the right direction.

Thanks in advance

BoschSensortec, 2018-05-14:
Hi @srf461,

Can you share your main code as a starting point?

srf461, 2018-05-14:
Of course, although there is nothing in there except for a routine where I was playing around with structures to figure out how they work and maybe get some insight into what I am doing wrong.
Here is my main.  It is called "Ta_main.c
/* 
 * File:   TA Main.c
 
 * Created on May 4, 2018, 6:12 PM
 */

#include <stdio.h> 
#include <stdlib.h>
#include "bme280.h"

/*
 * 
 */

void  main() 
{
/************************* TEST UTILIZATION OF STRUCTURE ************************/
    struct employee bob;
    struct employee susie;

    bob.userid = 1;
    susie.userid = 2;
    
    puts ("Enter the first name of employee 1");
    gets (bob.firstname);
    puts ("enter the first name of employee 2");
    gets (susie.firstname);
    
    printf("user 1 id is %d\n" , bob.userid);
    printf("user 2 first name is %s\n", susie. firstname);

    
    
    
/************************** END OF TEST STRUCTURE UTILIZATION ************************/    
    
}

In addition I defined the structure in bme280.h as follows:
/********************** DEFINING OF THE TEST STRUCTURE **************************/
struct employee
    {
        int userid;
        char firstname[25];
        char lastname[25];
        int age;
    };
/*********************** END OF STRUCTURE DEFINITION *************************/

But that's all there is.

srf461, 2018-05-16:
Hmmm, I was really hoping that you would reply.  Can you think of anything that I might be doing with regard to executing one of the bme280 functions in my main() program?

Really hoping to hear from you.

BoschSensortec, 2018-05-17:
Hi @srf461,

The basic information on how to get started is part of the [README](https://github.com/BoschSensortec/BME280_driver#example-for-spi-4-wire).  

As per the design, you do not need to modify the files provided in this repository until absolutely necessary as discussed earlier since most developers use the C99 standard and later.

You can have a look at our [gist](https://gist.github.com/BoschSensortec/cdaa6aa244335388d914de7f59443860) describing the implementation on the Arduino IDE. 

We currently do not have a standalone project for the PIC or other micro-controllers but we are working on this and would be releasing them during the course of this year.

srf461, 2018-05-17:
Thank you for your reply.  I can't remember whether I read the readme or not.  But I will do so now.

Thank you again.

srf461, 2018-05-17:
I had NOT read it and I am very embarrassed that I hadn't.  It looks like that file contains everything I need.
Thank you again for your patience with me.

srf461, 2018-05-18:
Yes, I do believe that a large part of my problem is that my compiler uses C89 and I am so uninformed about the C language that I am struggling to adapt your code to that standard.

srf461, 2018-07-06:
Update:
Due to my lack of understanding of the C language I was unable to figure out a lot of the Bosch code.  However, I did understand most of the BME280 data sheet.  So I wrote my own functions to perform the required operations.  My requirements for the BME280 device are very limited.  All I require is to initialize it once and then take readings and perform operations based on those readings.  The names that you used for your functions were very descriptive so I retained some of them although my code is markedly different, primarily because I do not understand STRUCTs.  Because of that I worked around it by declaring global variables to hold the “DIG”-type variable names as well as the compensated and uncompensated raw data.

My device is running in FORCED mode:

I wrote and have tested the following functions:
1.	void read_a_BME280_register(char x, char y, int z);      
2.	void write_a_BME280_register(char x, char y, char z);      
3.	void bme280_initialization();            
4.	void get_calib_data(); 
5.	void get_bme280_ctrl_hum();
6.	void get_bme280_ctrl_meas(); 
7.	void bme280_get_raw_sensr_data();
8.	void set_ctrl_hum_register();  
9.	void set_ctrl_meas_register();
10.	void set_config_register();
11.	void get_bme280_status();
12.	void compensate_temperatur();
13.	void compensate_humidit();
14.	void compensate_pressur();

All of the above functions are essentially shells that have the “read_a_BME280_register” function as an internal element.  The “shells” pass the control word address, the control word length, and the response length into the “read_a_BME280_register” function.  The function then writes the control word out to the BME device via the SPI bus and then provides the clock to read the BME’s response back in.  The received data is then written into the global "dig" type variables.   I figured this out from looking at the data sheet.  

I tested each of the functions 1 through 11 (BUT NOT THE THREE COMPENSATION FUNCTIONS) using a logic analyzer.  The calibration and raw sensor data is getting into the global variables as expected.  

I have started trying to figure out the compensation functions (items 12-14 above) but am not sure about what I am getting from them since I don’t understand the actual meanings of the raw data and calibration data.

I looked at the raw and calibration data that was provided by Tony_The_Lion in one of his posts.  Some of his data is exactly the same as mine.  However, other data elements are markedly different, and his pressure, temperature, and humidity data looks nothing like mine.  Now his was in decimal and mine is in hex in my registers, and I converted his to match mine.  As I said, some of it looks the same but others do not.  

I am including my global register data in the hope that you might be able to provide some insight.

I have high confidence in my functions that read the BME280 and write that data into the variables. 
There is a possibility that the operations that pack 8-bit data into 32 bit registers could be wrong.  I am currently looking at that to make sure I didn't transpose bytes somewhere.

And of course I have not begun looking at the compensation functions at all.  They do compile correctly, however, I have run only them and included the data for the sake of completeness.  

Item		Tony’s Data Decimal    	Hex	My data  (hex)
dig_T1:		28333			6ead		3742
dig_T2:		26573			67cd		32f3
dig_T3:		50			        32		        32
dig_P1:		37362			91f2		        47c5
dig_P2:		-10666			????		        6af2
dig_P3:		3024			bd0		        5d0
dig_P4:		7253			1c55		dd8
dig_P5:		50			        32    		25
dig_P6:		-7			        7ff9	        	7ff9
dig_P7:		9900			26ac		        180c
dig_P8:		-10230			d80a		68a0 (my conversion of his data is probably wrong)
dig_P9:		4285			10bd		0998
dig_H1:		75			        48    		4b
dig_H2:		352			        160	        	146
dig_H3:		0			         0		         0
dig_H4:		343			         157		         1a4
dig_H5:		0			         0          		32
dig_H6:		30			         1e    		1e
t_fine:		115103			1c19f		3d23c

My Uncompensated Temperature Data:			82693
My Compensated Temperature Data:			12b8

My Uncompensated Pressure Data:			529aa
My Compensated Pressure Data:			        1adb0	

My Uncompensated Humidity Data:			883e	
My Compensated Humidity Data:			        9f36

srf461, 2018-07-06:
The formatting of the data in my previous post did not survive the posting process, so i have attached the original document to this post.  It is a Microsoft Word document.
[github email.docx](https://github.com/BoschSensortec/BME280_driver/files/2171119/github.email.docx)

srf461, 2018-07-07:
Saturday Update:
At least part of the problem was what I suspected:  I was incorrectly packing the pressure and temperature calibration bytes.  I corrected that and the compensated temperature is now reading correctly.  I compared the reading against that of a digital thermometer.  The readings are almost identical.  Compensated humidity appears to be working as well, although i have no means of making a direct comparison.  
The only thing left is the pressure, which is stuck at 0x1ADB0 (110,000).  I am currently investigating that.

srf461, 2018-07-08:
Sunday Update:
I am still stuck on the pressure compensation portion of my program.  I have attached as a WORD file all that i have written below in case the formatting does not work once I transmit this post.

Due to my lack of understanding of the C language I was unable to figure out a lot of the Bosch code, particularly STRUCTs and the user-defined portion of the code that was in the README.MD file.  However, I did understand most of the BME280 data sheet.  So I wrote my own functions to perform the operations that I require.  

My requirements for the BME280 device are very limited.  All I need is to initialize the device once and then take periodic readings and perform operations based on those readings.  The names that Bosch used for the functions on Github are very descriptive so I retained some of them although my code is markedly different, primarily because I do not understand STRUCTs.  Due to my limitation in that area I worked around it by declaring global variables to hold the “DIG”-type variable names as well as the compensated and uncompensated raw data.

My device is running in FORCED mode.

I wrote and have tested the following functions:
1.	void read_a_BME280_register(char x, char y, int z); // this is a general purpose function that is called from within all functions that require data to be transmitted FROM the BME280

2.	void write_a_BME280_register(char x, char y, char z);   // this is a general purpose function that is called from within all functions that require data to be transmitted TO the BME280

3.	All of the functions that follow below are essentially shells that have the “read_a_BME280_register” and/or “write_a_BME280_register” functions as internal elements.  The “shells” pass the control word address, the control word length, & the response length into the “read_a_BME280_register” and/or “write_a_BME280_register”function as required.  The shell function then writes the control word out to the BME device via the SPI bus and then provides the clock to read the BME’s response back in.  I figured all of this out from looking at the data sheet.  

4.	void bme280_initialization(); // this function transmits the following two control words, one after the other:
a.	0x00 (receives a 0x00 response) 
b.	0x0d (receives a 0x60 response)

c.	void get_calib_data(); // 0x88 (receives 208 decimal bits in response)
i.	 The 208 bits include the following:
ii.	48 bits (6 bytes) of temperature cal data that is packed (in this function) into 32 bit words which are stored as dig_T1 through dig_T3
iii.	144 bits (18 bytes) of pressure cal data that is packed (in this function) into 32 bit words which are stored as dig_P1 through dig_P9
iv.	8 bits (1 byte) of data from register 0xa0 that will not be used
v.	8 bits (1 byte) of data from register 0xa1 that is packed and stored as dig_H1

vi.	This function also transmits 0xe1 (receives 64 bit response)
vii.	The 64 bits are packed and stored as dig_H2 through Dig_H6
        
1.	void set_ctrl_hum_register();  // this function transmits 0x01 (x1 humidity oversampling) to register 0x72. As part of the function it reads back the response from register 0xf2 and verifies the 0x01 response 

2.	void set_ctrl_meas_register(); // this function transmits 0xb5 (x16 temperature oversampling, x16 pressure oversampling, and “FORCED mode) to register 0x74. As part of the function it reads back the response from register 0xf4 and verifies the 0xb5 response 

3.	void set_config_register();  // this function transmits 0x10 (t_sb are DON’T CARE = 0, filter coefficient is 16, SPI 3-wire disabled) to register 0x75. As part of the function it reads back the response from register 0xf5 and verifies the 0x10 response 

All of the above function only run once.  The function below runs in a “while(1)” loop:

4.	void bme280_get_raw_sensr_data();  // this function transmits 0xf7 and receives a 64 bit response
a.	The 64 bits include:
i.	Three bytes which are packed and stored as uncompensated pressure (u_pressure)
ii.	Three bytes which are packed and stored as uncompensated temperature (u_temperature)
iii.	Two bytes which are packed and stored as uncompensated humidity (u_humidity)

All of the above have been tested and verified against my logic analyzer which is monitoring the SPI bus /CSB, SCK, “Data to BME” and “Data From BME” signals.
After running my board with the BME280 device and putting a breakpoint after the raw data is stored I have the following:

My calibration data is as follows:
Item		My data  (hex)
dig_T1:		6e42
dig_T2:		65f3
dig_T3:		0032
dig_P1:		8ec5
dig_P2:		d5f2
dig_P3:		0bd0
dig_P4:		1bd8
dig_P5:		0025
dig_P6:		fff9
dig_P7:		300c
dig_P8:		d120
dig_P9:		1388
dig_H1:		4b
dig_H2:		0146
dig_H3:		0
dig_H4:		01a4
dig_H5:		32
dig_H6:		1e
t_fine:		1cff6

My Uncompensated Temperature Data:		8075c	
[... 39 lines cut ...]
        var5 = (1048576) - u_pressure;
        c_pressure = ((var5 - (var2 / 4096))) * 3125;
        if (c_pressure < 0x80000000)
            c_pressure = (c_pressure << 1) / (var1);
        else
            c_pressure = (c_pressure / var1) * 2;
            var1 = ((dig_P9) * ((((c_pressure / 8) * (c_pressure / 8)) / 8192))) / 4096;
            var2 = (((c_pressure / 4)) * (dig_P8)) / 8192;
            c_pressure = (c_pressure + ((var1 + var2 + dig_P7) / 16));
    
        if (c_pressure < pressure_min)
            c_pressure = pressure_min;
        else if (c_pressure > pressure_max)
            c_pressure = pressure_max;
    } 
    else 
    {
        c_pressure = pressure_min;
    }
}
I would appreciate an opinion regarding what might be wrong with this function.
Sincerely,
Ray Russell 
Houston TX USA
[github email 2.docx](https://github.com/BoschSensortec/BME280_driver/files/2173778/github.email.2.docx)

srf461, 2018-07-09:
Good morning and thank you for your reply.
As I understand it there are three versions of the "compensate_pressure" function.  In order, they are:
1) the floating point version, 2) the 64-bit version, and 3) the 32-bit version.

The limits for the floating point and the 32-bit version appear to be the same.  The version that I copied from my code and pasted into my post above is I believe the 32-bit version.  Is that incorrect?

srf461, 2018-07-09:
The pressure calibration values that have been received from my device and stored are as enumerated above.
I started my program running in debug mode.  I stopped it randomly and inserted a breakpoint at the end of the humidity compensation program.  That is the point where the calib values have been read, packed and stored, the raw data has been read, packed and stored, and the temperature, pressure, and humidity data has been compensated.  The data values are as follows:

Un-calibrated pressure reading (u_pressure) is: 0x5298c
The t_fine value associated with the u_pressure reading was: 0x217b6
The compensated pressure reading (c_pressure) is: 0x20f2e which corresponds to 1349.58 

I have tried to make sure my typedefs are correct and that I am packing the raw and cal bytes properly.  I am out of ideas now.

Could you possibly plug my values into your program and see if you get the same result?

Thank you

BoschSensortec, 2018-07-10:
Hi @srf461,

The 64bit was recommended to check if you would have the same behavior as with the 32bit which doesn't seem the case. In the worst case it could be how the compiler interprets the operations and performs them. 

The original code [here](https://github.com/BoschSensortec/BME280_driver/blob/cf40d00b0b5139e287b670881c433c0041d98d9f/bme280.c#L2123) uses shifts for multiplications or divisions. This should work as a last resort.

srf461, 2018-07-10:
The 64-bit hits the minimum stop while the 32-bit hits the maximum stop.  I will try it now and hopefully get back to you before you leave for the day.

srf461, 2018-07-10:
There are constants in this code that I don't see anywhere else.  

For the first one:  "BME280_INIT_VALUE"-Is this just the uncompensated pressure obtained from reading the raw sensor data or something else?

There are quite a few constants like "BME280_SHIFT_BIT_POSITION_BY_17_BITS" which in this case I assume I could just replace it with "17"?

My IDE editor is choking on the "s64" typedef.  Is there a way around this or might there be a 32 bit version of this function that uses shifts for multiplications and divisions?

BoschSensortec, 2018-07-10:
Hi @srf461,

Due to previously poor code implementation we have strange constants like `BME280_INIT_VALUE` which is actually defined as 0, `BME280_SHIFT_BIT_POSITION_BY_17_BITS` is actually the value 17. But do refer the bme280.h file to be sure.

You can replace the types like below,
```c
u8 uint8_t
s8 int8_t
u16 uint16_t
s16 int16_t
u32 uint32_t
s32 int32_t
u64 uint64_t
s64 int64_t
``` 

If you compiler doesn't support int64_t use the 32bit version instead.

srf461, 2018-07-10:
Thank you for the very quick reply.  I will do as you suggested and let you know how it goes.

srf461, 2018-07-10:
I replaced my original pressure compensation code with the version of the code that you sent/pointed to.  I modified it to remove the returns and to replace the typedefs as you suggested.  My compiler does indeed support int64_t and it compiled successfully.

I had previously "forced" the program to hold the u_pressure, and t_fine constant, and I ran this version that way as well.  Here is my code and the data that I used is just below that:

void compensate_pressur()
{
	int64_t v_x1_s64r = 0;
	int64_t v_x2_s64r = 0;
	int64_t c_pressure = 0;

	v_x1_s64r = ((int64_t)t_fine) - 128000;
	v_x2_s64r = v_x1_s64r * v_x1_s64r * (int64_t)dig_P6;
	v_x2_s64r = v_x2_s64r + ((v_x1_s64r * (int64_t)dig_P5) << 17);
	v_x2_s64r = v_x2_s64r + (((int64_t)dig_P4) << 35);
	v_x1_s64r = ((v_x1_s64r * v_x1_s64r * (int64_t)dig_P3) >> 8) + ((v_x1_s64r * (int64_t)dig_P2) << 12);
	v_x1_s64r = (((((int64_t)1) << 47) + v_x1_s64r)) * ((int64_t)dig_P1) >> 33;
	c_pressure = 1048576 - u_pressure;
	/* Avoid exception caused by division by zero */
	if (v_x1_s64r != 0)
		#if defined __KERNEL__
			c_pressure = div64_s64((((c_pressure << 31) - v_x2_s64r) * 3125), v_x1_s64r);
		#else
			c_pressure = (((c_pressure << 31) - v_x2_s64r) * 3125) / v_x1_s64r;
		#endif
	else
//		return BME280_INVALID_DATA;
	v_x1_s64r = (((int64_t)dig_P9) * (c_pressure >> 13) * (c_pressure >> 13)) >> 25;
	v_x2_s64r = (((int64_t)dig_P8) * c_pressure) >> 19;
	c_pressure = (((c_pressure + v_x1_s64r + v_x2_s64r) >> 8) + (((int64_t)dig_P7) << 4));

}

u_pressure (held constant)-- x052d88
t_fine (held constant)-- 115103 (decimal)
c_pressure result:  0x1a6551

I then allowed it to run without constraints.  A typical sample is as follows:
u_pressure:  0x5272f
t_fine: 0x1fd12
c_pressure:  0x5b9c69 = 6003817

I am very confused.  I really hate tying you up with this problem and would like to try and evaluate it myself.  However, I have no idea of what the uncompensated pressure actually is and how the various calibration parameters affect it.  Is there any public documentation that explains it?

srf461, 2018-07-10:
One more thing:  When I ran the code I placed a breakpoint at the "return BME280_INVALID_DATA" exception.  It never executed that code.  So it appears to be running as expected even if it is not providing the expected result.

BoschSensortec, 2018-07-10:
Hi @srf461,

To answer your question in brief, the uncompensated values are related to the pressure measured. Due to the uniqueness of each sensor, the deviation is stored in the form of the calibration parameters. Hence the combination of the 2 provides you with the compensated pressure. The only documentation publicly available is in the [datasheet](https://ae-bst.resource.bosch.com/media/_tech/media/datasheets/BST-BME280_DS001-12.pdf) under Chapter 4.

Also, you need to comment out the `else` statement or the code will perform 1 less operation.

Assuming you are triggering (forcing) multiple measurements and that the values of uncompensated pressure and temperature are changing the only problem would be with how the compiler interprets this code and this might take some time for us to setup and debug.

Replace the pressure compensation with the snippet below

```c
void compensate_pressur()
{
	int64_t var1 = 0;
	int64_t var2 = 0;
	int64_t c_pressure = 0;

	var1 = ((int64_t)t_fine) - 128000;
	var2 = var1 * var1 * (int64_t)dig_P6;
	var2 = var2 + ((var1 * (int64_t)dig_P5) << 17);
	var2 = var2 + (((int64_t)dig_P4) << 35);
	var1 = ((var1 * var1 * (int64_t)dig_P3) >> 8) + ((var1 * (int64_t)dig_P2) << 12);
	var1 = (((((int64_t)1) << 47) + var1)) * ((int64_t)dig_P1) >> 33;
	
	if (var1 != 0)
		c_pressure = (((c_pressure << 31) - var2) * 3125) / var1;
	else 
		return;

	c_pressure = 1048576 - u_pressure;
	var1 = (((int64_t)dig_P9) * (c_pressure >> 13) * (c_pressure >> 13)) >> 25;
	var2 = (((int64_t)dig_P8) * c_pressure) >> 19;
	c_pressure = (((c_pressure + var1 + var2) >> 8) + (((int64_t)dig_P7) << 4));
}
```

srf461, 2018-07-10:
/* To answer your question in brief, the uncompensated values are related to the pressure measured. Due to the uniqueness of each sensor, the deviation is stored in the form of the calibration parameters. Hence the combination of the 2 provides you with the compensated pressure. The only documentation publicly available is in the datasheet under Chapter 4. */

Yes, I do understand that the calibration values that are returned from the sensor are unique for that sensor.  My effort to hold everything constant meant that the only things i did were to commented out the line of the get raw data function that provides the un-compensated data.  Then I went to the typedefs section and forced "u_pressure" to be the value that I wanted.  I think I sent you that portion of the code.

Then I did the same thing in the "compensate_temperatur()" function, where I commented out the line that saves the "t_fine" variable.  Then I again went to my typedefs section and forced "t_fine" to be what I wanted.  Those were the only things i did.

/* Also, you need to comment out the else statement or the code will perform 1 less operation. */
I did that before I pasted the snippet in.  The resultant c_pressure became 0xffffffff for some reason.  i did not investigate, I just put the code back in minus the line where it returned the error code.

/*Assuming you are triggering (forcing) multiple measurements and that the values of uncompensated pressure and temperature are changing the only problem would be with how the compiler interprets this code and this might take some time for us to setup and debug.*/
I agree, that's why I forced the u_pressure and t_fine variables earlier--to hold everything steady.  I have removed those "forces" by commenting them out.  They are available to put back in when/if it makes sense to do so.

/* Replace the pressure compensation with the snippet below */
Done.  i dropped the snippet in without modification.  I allowed it to run and after a few seconds i inserted a breakpoint to stop execution at the end of the compensate_pressure function.  The results are:
u_pressure 0x529a5
t_fine          0x21209
c_pressure 0x275a0 = 161184 decimal

I then removed the breakpoint to allow it to run free for several seconds.  Then I inserted a breakpoint in the same place.  Results:
u_pressure 0x52a23
t_fine          0x214ed
c_pressure 0x2759f = 161183 decimal

I then did it a third time.  Results:
u_pressure 0x52a37
t_fine          0x2152c
c_pressure 0x2759f = 161183 decimal

And one last time.  Results:
u_pressure 0x52b1a
t_fine          0x219b4
c_pressure 0x2759e = 161182 decimal

So the results from sample to sample are very consistent.  I think they are just wrong!  Lol

srf461, 2018-07-10:
I started going through the code line by line and doing the arithmetic manually.  I found that line 4 (with the 35 bit shift) produces a result that is truncated at 32 bits.  I was not able to figure out how to get around it.  I went out to the chip manufacturer's forum and it seems that there is no native support for values greater than 32 bit even though the compiler does not choke on 64 bit declarations.

I'm going to fall back and do the same thing with the 32 bit integer function.

BoschSensortec, 2018-07-11:
Hi @srf461,

Which line in particular was it being truncated at?

srf461, 2018-07-11:
Just in case you care to use them:
These are Tony_The_Lion's calibration numbers.  I was using them back when I was not confident that my packed cal numbers were correct.  I am confident now, I have just kept using these.  Once I get through my evaluation of the 32 compensation function I will remove them and allow my functions to fill them in.
But as I said for the moment these are the ones that I am using.

u_pressure   0x529a5
t_fine           115103 decimal
dig_P1          37362
dig_P2         -10666
dig_P3         3024
dig_P4         7253
dig_P5         50
dig_P6         -7
dig_P7         9900
dig_P8         -10230
dig_P9         4285

srf461, 2018-07-12:
I have compared the numbers generated by the program against values that I have obtained by performing the same computations on a calculator.  There are some differences, most likely in the negative numbers.  I will look at them in the morning.  In the meantime is it possible for you to plug my u_pressure and pressure calibration numbers into your version of the 32-bit integer pressure compensation function?  That would tell me whether the problem is in the way my compiler is handling the function itself or whether I am putting bad numbers into it.  

The "c_pressure" value that I get at the end of the function is 0x18266 or 98,918 decimal.

BoschSensortec, 2018-07-12:
Hi @srf461,

989.18hPa is a valid measurement. We will test out your request and get back to you.

srf461, 2018-07-12:
Actually, let me un-modify my code so that it will use my own parameters.  I will send them to you in a couple of minutes.

srf461, 2018-07-12:
MY REAL DATA:
dig_P1:     0x8ec5
dig_P2:     0xd5f2
dig_P3:     0x0bd0
dig_P4:     0x1bd8
dig_P5:     0x0025
dig_P6:     0xfff9
dig_P7:     0x300c
dig_P8:     0xd120
dig_P9:     0x1388
Several t_fine and c_pressure readings:
u_pressure     t_fine       c_pressure
0x52ed6       0x1f8be            0x18d4c    1017.08
0x52cf8        0x1edc9            0x18d48    1017.04
0x52be8       0x1e874            0x18d48    1017.04
0x52b41       0x1e51f             0x18d4c    1017.08
Current local baro pressure per the internet:  30.04 inHg/1 017.271 hPa as of 6:15am local time

As you can see the numbers look like things are working.  The one thing I did not do prior to forcing Tony_The_Lion's numbers into my functions was to actually run it see whether it was working.  I should have done that first.
However, I would still appreciate it if you would plug my cal values and at least one of my t_fine and u_press values in to cross-check me.

ma-harvey, 2018-07-12:
@srf461 : I can share with you an excel spreadsheet we use inside Bosch to cross-validate the implementation of the C function. Our company network blocks the uploading of files here, send me a private message with your email address and I can share with you.

srf461, 2018-07-13:
Apologies, but although I have looked I don't see any way to send you a private message.  I will respond with my email address if you can show me the way to do it.

srf461, 2018-07-13:
i don't know whether this will get kicked or not but you can send it to bit.buckets57   AT   gmail dot com.

srf461, 2018-07-19:
Apologies for taking so long to get back but the excel file that was sent by email worked perfectly.  Thanks to you for taking the time to help me.

Ray Russell

srf461, 2018-07-23:
Hello all,
Marc-Andre Harvey, Bosch's US Field Application Engineer asked me to share with you a spreadsheet that they created to accept the uncompensated BME280 temperature, pressure, and humidity data along with the calibration data from the device registers.  The spreadsheet returns all of the variables (vars) that are computed in the T, P, and H compensation functions as well as Tfine which is computed in the Temperature compensation function and is passed into and used by the Pressure compensation function.  It works well and allowed me to validate my data against the data that they believe i should get. 

Many thanks to the Bosch team--You went above and beyond the call of duty for me and I do appreciate it.
Ray Russell
Houston, TX
[sample_calculation_BME280.xlsx](https://github.com/BoschSensortec/BME280_driver/files/2219741/sample_calculation_BME280.xlsx)

ma-harvey, 2018-11-19:
As I understand the issue has been resolved. 
@srf461 : Would you be kind enough to mark the issue as closed ?