boschsensortec/BME280_SensorAPI #52: Templates for function pointers

a5021, 2018-06-03:
The type of last parameter (**len**) of user's read and wirte functions is **uint16_t**. BME280 has only 45 registers one byte each. What those 16-bits are for ? Is there some secret method to read 65535 bytes from BME280 or it is a bug in the code ?

ma-harvey, 2018-06-04:
As the size of the FIFO for most of our parts is larger than 255 bytes, we have move the read and write APIs to use 16 bits variables for compatibility, so the same function can be used to read the data from your pressure sensors and for your IMU.

Even our newest pressure sensor, BMP388, has a 512 bytes FIFO.

a5021, 2018-06-04:
Thank you for your explanations. BMP280 and BME280 have no FIFO but API changed because BMP388 has. Looks a bit strange.

closed 2018-06-04 by a5021