boschsensortec/BME280_SensorAPI #53: Error on reading values

adanielc, 2018-06-12:
I'm trying to read from a Sam d21 Xplained pro but wvery time I got the same read 0,3000,0

there's my code:

#include <asf.h>
#include <string.h>
#include <stdio.h>

#include "bme280.h"

// PREPROCESSOR CONSTANTS
#define BAUDRATE 115200
#define SLAVE_ADDRESS 0x76
#define BME_ID 0x60
#define TIMEOUT 1000

// STRUCTURES
struct usart_module uart_instance;
struct i2c_master_module i2c_master_instance;
struct bme280_dev dev;
struct bme280_data comp_data;

// PROTOTYPES
void print_sensor_data(struct bme280_data *comp_data);
void handle_bme280_error(int8_t _rslt);
void uart_setup(void);
void i2c_master_setup(void);
void user_delay_ms(uint32_t period);
int8_t user_i2c_read(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t len);
int8_t user_i2c_write(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t len);
int8_t stream_sensor_data_normal_mode(struct bme280_dev *dev);
int32_t temp;

int main (void)
{
	system_init();
	uart_setup();
	i2c_master_setup();
	
	    struct bme280_dev dev;
	    struct bme280_data comp_data;
	    int8_t rslt = BME280_OK;

	    dev.dev_id = BME280_I2C_ADDR_PRIM;
	    dev.intf = BME280_I2C_INTF;
	    dev.read = user_i2c_read;
	    dev.write = user_i2c_write;
	    dev.delay_ms = user_delay_ms;

	   rslt = bme280_init(&dev);
	   handle_bme280_error(rslt);

		rslt = stream_sensor_data_normal_mode(&dev);
		handle_bme280_error(rslt);

	while (1) 
	{
		
		temp=comp_data.temperature;
		
	}
	 
}

// UART SETUP
void uart_setup(void){
	
	struct usart_config uart_configuration;
	
	usart_get_config_defaults(&uart_configuration);
	
	uart_configuration.baudrate = BAUDRATE;
	uart_configuration.mux_setting = EDBG_CDC_SERCOM_MUX_SETTING;
	uart_configuration.pinmux_pad0 = EDBG_CDC_SERCOM_PINMUX_PAD0;
	uart_configuration.pinmux_pad1 = EDBG_CDC_SERCOM_PINMUX_PAD1;

	
	stdio_serial_init(&uart_instance, EDBG_CDC_MODULE, &uart_configuration);
	usart_enable(&uart_instance);
}

// I2C CONFIGURATION:
void i2c_master_setup(void) {
	
	struct i2c_master_config i2c_master_configuration;
	
	//Activar las resistencias internas
	//port_pin_set_config(PIN_PA08,PORT_PIN_PULL_UP);
	//port_pin_set_config(PIN_PA09,PORT_PIN_PULL_UP);
	
	/* Configure OVERFLOW callback */
	i2c_master_get_config_defaults(&i2c_master_configuration);
	
	i2c_master_configuration.generator_source = GCLK_GENERATOR_0;
	i2c_master_configuration.baud_rate = I2C_MASTER_BAUD_RATE_100KHZ;
	i2c_master_configuration.buffer_timeout = 10000;
	i2c_master_configuration.pinmux_pad0 = PINMUX_PA08C_SERCOM0_PAD0;
	i2c_master_configuration.pinmux_pad1 = PINMUX_PA09C_SERCOM0_PAD1;
	
	
	
	while(i2c_master_init(&i2c_master_instance, SERCOM0, &i2c_master_configuration) != STATUS_OK);
	i2c_master_enable(&i2c_master_instance);
}

int8_t stream_sensor_data_normal_mode(struct bme280_dev *dev)
{
	int8_t rslt;
	uint8_t settings_sel;
	struct bme280_data comp_data;

	/* Recommended mode of operation: Indoor navigation */
	dev->settings.osr_h = BME280_OVERSAMPLING_1X;
	dev->settings.osr_p = BME280_OVERSAMPLING_16X;
	dev->settings.osr_t = BME280_OVERSAMPLING_2X;
	dev->settings.filter = BME280_FILTER_COEFF_16;
	dev->settings.standby_time = BME280_STANDBY_TIME_62_5_MS;

	settings_sel = BME280_OSR_PRESS_SEL;
	settings_sel |= BME280_OSR_TEMP_SEL;
	settings_sel |= BME280_OSR_HUM_SEL;
	settings_sel |= BME280_STANDBY_SEL;
	settings_sel |= BME280_FILTER_SEL;
	rslt = bme280_set_sensor_settings(settings_sel, dev);
	rslt = bme280_set_sensor_mode(BME280_NORMAL_MODE, dev);

	printf("Temperature, Pressure, Humidity\r\n");
	/* Delay while the sensor completes a measurement */
	dev->delay_ms(70);
	rslt = bme280_get_sensor_data(BME280_ALL, &comp_data, dev);
	print_sensor_data(&comp_data);

	return rslt;
}

void print_sensor_data(struct bme280_data *comp_data)
{
	#ifdef BME280_FLOAT_ENABLE
	printf("%0.2f, %0.2f, %0.2f\r\n",comp_data->temperature, comp_data->pressure, comp_data->humidity);
	#else
	printf("%ld, %ld, %ld\r\n",comp_data->temperature, comp_data->pressure, comp_data->humidity);
	#endif
}

void handle_bme280_error(int8_t _rslt)
{
	if (_rslt < BME280_OK) {
		printf("BME280 Error code: %d", _rslt);
		for(;;); /* Trap if error */
		} else if (_rslt > BME280_OK) {
			
		printf("BME280 Warning code: %d", _rslt);
	}
}

void user_delay_ms(uint32_t period)
{
	
	

}

int8_t user_i2c_read(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t len)
{
    int8_t rslt = 0; /* Return 0 for Success, non-zero for failure */
	
	int16_t timeout = 0;
	
	/* Initialize i2c packet. */
	struct i2c_master_packet packet = {
		.address     = &reg_addr,
		.data_length = len,
		.data        = reg_data,
		.ten_bit_address = false,
		.high_speed      = false,
		.hs_master_code  = 0x0,
	};
	
	//Read packet
	while (i2c_master_read_packet_wait(&i2c_master_instance, &packet) != STATUS_OK) {
		if (timeout++ == TIMEOUT) {	return false; }	// Abort if timeout
	}

    return rslt;
}

int8_t user_i2c_write(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t len)
{
    int8_t rslt = 0; /* Return 0 for Success, non-zero for failure */
	int16_t timeout = 0;
	
	/* Initialize i2c packet. */
	struct i2c_master_packet packet = {
		.address     = &reg_addr,
		.data_length = sizeof(reg_data),
		.data        = reg_data,
		.ten_bit_address = false,
		.high_speed      = false,
		.hs_master_code  = 0x0,
	};
	
	// Send packet:
	/* Write buffer to slave until success. SEND REGISTER WE WANT TO READ */
	while (i2c_master_write_packet_wait_no_stop(&i2c_master_instance, &packet) != STATUS_OK) {
		if (timeout++ == TIMEOUT) {	return false; }	// Abort if timeout
	}

    return rslt;
}

BoschSensortec, 2018-06-13:
@adanielc,

You need to provide a delay function for the driver to function properly. You have currently linked it to a dummy function.

adanielc, 2018-06-13:
Thanks! I'll add the Delay(perdiod) inside the delay function and let you know.

adanielc, 2018-06-13:
I'm still having problems with the delay, I changed the "dev->delay_ms"  to just the delay_ms and I'm still getting the same values 0,30000,0

/**
 * \file
 *
 * \brief Empty user application template
 *
 */

#include <asf.h>
#include <string.h>
#include <stdio.h>
#include "bme280.h"
#include <delay.h>

// PREPROCESSOR CONSTANTS
#define BAUDRATE 115200
#define SLAVE_ADDRESS 0x76
#define BME_ID 0x60
#define TIMEOUT 1000

// STRUCTURES
struct usart_module uart_instance;
struct i2c_master_module i2c_master_instance;
struct bme280_dev dev;
struct bme280_data comp_data;

// PROTOTYPES
void print_sensor_data(struct bme280_data *comp_data);
void handle_bme280_error(int8_t _rslt);
void uart_setup(void);
void i2c_master_setup(void);
void user_delay_ms(uint32_t period);
int8_t user_i2c_read(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t len);
int8_t user_i2c_write(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t len);
int8_t stream_sensor_data_normal_mode(struct bme280_dev *dev);
int32_t temp;

int main (void)
{
	system_init();
	uart_setup();
	i2c_master_setup();
	
	    int8_t rslt = BME280_OK;

	    dev.dev_id = BME280_I2C_ADDR_PRIM;
	    dev.intf = BME280_I2C_INTF;
	    dev.read = user_i2c_read;
	    dev.write = user_i2c_write;
	   

			
	     rslt = bme280_init(&dev);
	     handle_bme280_error(rslt);
		 rslt = stream_sensor_data_normal_mode(&dev);
	     handle_bme280_error(rslt);
		

	while (1) 
	{
		
		
	}
	 
}

// UART SETUP
void uart_setup(void){
	
	struct usart_config uart_configuration;
	
	usart_get_config_defaults(&uart_configuration);
	
	uart_configuration.baudrate = BAUDRATE;
	uart_configuration.mux_setting = EDBG_CDC_SERCOM_MUX_SETTING;
	uart_configuration.pinmux_pad0 = EDBG_CDC_SERCOM_PINMUX_PAD0;
	uart_configuration.pinmux_pad1 = EDBG_CDC_SERCOM_PINMUX_PAD1;

	
	stdio_serial_init(&uart_instance, EDBG_CDC_MODULE, &uart_configuration);
	usart_enable(&uart_instance);
}

// I2C CONFIGURATION:
void i2c_master_setup(void) {
	
	struct i2c_master_config i2c_master_configuration;
	
	//Activar las resistencias internas
	//port_pin_set_config(PIN_PA08,PORT_PIN_PULL_UP);
	//port_pin_set_config(PIN_PA09,PORT_PIN_PULL_UP);
	
	/* Configure OVERFLOW callback */
	i2c_master_get_config_defaults(&i2c_master_configuration);
	
	i2c_master_configuration.generator_source = GCLK_GENERATOR_0;
	i2c_master_configuration.baud_rate = I2C_MASTER_BAUD_RATE_100KHZ;
	i2c_master_configuration.buffer_timeout = 10000;
	i2c_master_configuration.pinmux_pad0 = PINMUX_PA08C_SERCOM0_PAD0;
	i2c_master_configuration.pinmux_pad1 = PINMUX_PA09C_SERCOM0_PAD1;
	
	
	
	while(i2c_master_init(&i2c_master_instance, SERCOM0, &i2c_master_configuration) != STATUS_OK);
	i2c_master_enable(&i2c_master_instance);
}

int8_t stream_sensor_data_normal_mode(struct bme280_dev *dev)
{
	int8_t rslt;
	uint8_t settings_sel;
	struct bme280_data comp_data;

	/* Recommended mode of operation: Indoor navigation */
	dev->settings.osr_h = BME280_OVERSAMPLING_1X;
	dev->settings.osr_p = BME280_OVERSAMPLING_16X;
	dev->settings.osr_t = BME280_OVERSAMPLING_2X;
	dev->settings.filter = BME280_FILTER_COEFF_16;
	dev->settings.standby_time = BME280_STANDBY_TIME_62_5_MS;

	settings_sel = BME280_OSR_PRESS_SEL;
	settings_sel |= BME280_OSR_TEMP_SEL;
	settings_sel |= BME280_OSR_HUM_SEL;
	settings_sel |= BME280_STANDBY_SEL;
	settings_sel |= BME280_FILTER_SEL;
	rslt = bme280_set_sensor_settings(settings_sel, dev);
	rslt = bme280_set_sensor_mode(BME280_NORMAL_MODE, dev);

	printf("Temperature, Pressure, Humidity\r\n");
	/* Delay while the sensor completes a measurement */
	delay_ms(70);
	rslt = bme280_get_sensor_data(BME280_ALL, &comp_data, dev);
	print_sensor_data(&comp_data);

	return rslt;
}

void print_sensor_data(struct bme280_data *comp_data)
{
	//#ifdef BME280_FLOAT_ENABLE
	//printf("%0.2f, %0.2f, %0.2f\r\n",comp_data->temperature, comp_data->pressure, comp_data->humidity);
//	#else
	printf("%ld, %ld, %ld\r\n",comp_data->temperature, comp_data->pressure, comp_data->humidity);
	//#endif
}

void handle_bme280_error(int8_t _rslt)
{
	if (_rslt < BME280_OK) {
		printf("BME280 Error code: %d", _rslt);
		for(;;); /* Trap if error */
		} else if (_rslt > BME280_OK) {
			
		printf("BME280 Warning code: %d", _rslt);
	}
}

//void user_delay_ms(uint32_t period)
//{
	
//	delay_ms(period);

// }

int8_t user_i2c_read(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t len)
{
    int8_t rslt = 0; /* Return 0 for Success, non-zero for failure */
	int16_t timeout = 0;
	
	/* Initialize i2c packet. */
	struct i2c_master_packet packet = {
		.address     = &reg_addr,
		.data_length = len,
		.data        = reg_data,
		.ten_bit_address = false,
		.high_speed      = false,
		.hs_master_code  = 0x0,
	};
	
	//Read packet
	while (i2c_master_read_packet_wait(&i2c_master_instance, &packet) != STATUS_OK) {
		if (timeout++ == TIMEOUT) {	return false; }	// Abort if timeout
	}

    return rslt;
}

int8_t user_i2c_write(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t len)
{
    int8_t rslt = 0; /* Return 0 for Success, non-zero for failure */
	int16_t timeout = 0;
	
	/* Initialize i2c packet. */
	struct i2c_master_packet packet = {
		.address     = &reg_addr,
		.data_length = sizeof(reg_data),
		.data        = reg_data,
		.ten_bit_address = false,
		.high_speed      = false,
		.hs_master_code  = 0x0,
	};
	
	// Send packet:
	/* Write buffer to slave until success. SEND REGISTER WE WANT TO READ */
	while (i2c_master_write_packet_wait_no_stop(&i2c_master_instance, &packet) != STATUS_OK) {
		if (timeout++ == TIMEOUT) {	return false; }	// Abort if timeout
	}

    return rslt;
}

BoschSensortec, 2018-06-20:
Hi @adanielc,

Refer to this [gist](https://gist.github.com/BoschSensortec/5ff33891ae823ef9288666b8d915c09b).

flo90, 2018-06-23:
Hi @adanielc,
just saw this thread as i had a similar issue.
Your approach with the delay_ms(...) inside your user_delay_ms(...) was correct, but using ASF you have to initialize delay_ms(...) with delay_init() at the beginning.

adanielc, 2018-06-25:
I forgot to thank you! it work perfect. my error was on Write/Read function. it works perfect know.