boschsensortec/BME280_SensorAPI #59: ESP-IDF

klapligehesten, 2018-08-20:
Hi,

I am using your code in a espressif ESP-IDF interface on esp32 chip and the I2C interface.
Everything compiles ok, but for some reason the bme280_init functions:
&nbsp;&nbsp;&nbsp;bme280_soft_reset 
and 
&nbsp;&nbsp;&nbsp;get_calib_data 
requires a small delay between them to get the calib_data structure initiated.
```
....
/* Reset the sensor */
rslt = bme280_soft_reset(dev);
if (rslt == BME280_OK) {
	/* Read the calibration data */
	dev->delay_ms(40); // My 'hack'
	rslt = get_calib_data(dev);
}
break;
....
```
For now, I have made a work around by delaying 10ms my dev->write assigned function.
It's not nice but it does the trick.
  
This is just a remark, but if you have an explanation i'll be pleased to hear it :-)

BoschSensortec, 2018-08-21:
Hi @klapligehesten,

Some users have reported that the sensor has not yet booted up after the soft-reset.

One solution would be that the [delay](https://github.com/BoschSensortec/BME280_driver/blob/a50e719dedec9c08efc3c8a9708881dfc4940848/bme280.c#L576) in the `bme280_soft_reset` function may need to be increased depending on your system.  The start-up time mentioned in the datasheet is 2ms and this is what is expected when the `delay_ms` function pointer is called. 
The other solution, if you do not want to modify the code would be that the `delay_ms` function linked provide more delay than requested.

klapligehesten, 2018-08-21:
Hi,
Thanks for the answer, and the explanation that the sensor has not booted yet is very plausible.
I have tested the code increasing the dev->delay_ms(2) in the bme280_soft_reset from 3ms and up to 10ms. And first on 10ms my program started to report the correct values. So i am not sure sure that its another problem in my application.
The code you have written is nice and generic and compiled the first time on the esp-idf platform.
I trust that the datasheet soft reset start-up time 2ms is correct and the problem is related to the platform/my program and where it is used (inaccuracy in ms timing of delay, communication code etc..).
So my opinion is not to change any of your code and close this issue.
At most make a remark on the start-up timing in the readme.
Cheers :-)

closed 2018-08-21 by klapligehesten