boschsensortec/BME280_SensorAPI #67: Cannot interface BME280 and nRF52382 (SPI)

ASL07, 2018-12-21:
I have posted the question on BME160's Github, just paste it here for more visibility and because this is the BME280_driver github

Hi @BoschSensortec @XaviLC,

Your comments regarding configuration of the sensor have been very helpful, however I still cannot get the BME280 to work with the nRF52832 (I am using the RAK8211-NB board from RAK wireless).

After running rslt = bme280_init(&dev); , rslt is empty, and, in the function bme280_get_regs(), I get rslt = BME280_E_COMM_FAIL when trying to read BME280_CHIP_ID_ADDR.

Could you share your complete .c file with the spi instance and sensor initialization/read code?

I share my code below and hope you can see what I am missing.

#include "nrf_drv_spi.h"
#include "app_util_platform.h"
#include "nrf_gpio.h"
#include "nrf_delay.h"
#include "boards.h"
#include "app_error.h"
#include <string.h>
#include "nrf_log.h"
#include "nrf_log_ctrl.h"
#include "nrf_log_default_backends.h"
#include "bme280.h"

#define SPI_INSTANCE  0 /**< SPI instance index. */
#define SPI_SS_PIN 2
#define SPI_MISO_PIN 3
#define SPI_MOSI_PIN 5
#define SPI_SCK_PIN 4

static const nrf_drv_spi_t spi = NRF_DRV_SPI_INSTANCE(SPI_INSTANCE);

int8_t spi1_read_transfer(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t length)
{
	ret_code_t ret;
	
        nrf_gpio_pin_clear(SPI_SS_PIN); /* To activate the sensor on the bus, i.e., set the chip/slave select low */

	
	ret = nrf_drv_spi_transfer(&spi, &reg_addr, 1, NULL, 0);
	if(ret == NRF_SUCCESS)
		ret = nrf_drv_spi_transfer(&spi, NULL, 0, reg_data, length);
	
	nrf_gpio_pin_set(SPI_SS_PIN); /* To activate the sensor on the bus, i.e., set the chip/slave select high */
	
	return (int8_t)ret;	
}

int8_t spi1_write_transfer(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t length)
{
	ret_code_t ret;
	
	nrf_gpio_pin_clear(SPI_SS_PIN); /* To activate the sensor on the bus, i.e., set the chip/slave select low */
	
	ret = nrf_drv_spi_transfer(&spi, &reg_addr, 1, NULL, 0);
	if(ret == NRF_SUCCESS)
		ret = nrf_drv_spi_transfer(&spi, reg_data, length, NULL, 0);
	
	nrf_gpio_pin_set(SPI_SS_PIN); /* To activate the sensor on the bus, i.e., set the chip/slave select high */

	
	return (int8_t)ret;	
}

void user_delay_ms(uint32_t period)
{
    /*
     * Return control or wait,
     * for a period amount of milliseconds
     */
     nrf_delay_ms(period);
}

int8_t stream_sensor_data_forced_mode(struct bme280_dev *dev) 
{
    int8_t rslt;
    uint8_t settings_sel;
    struct bme280_data comp_data;

    /* Recommended mode of operation: Indoor navigation */
    dev->settings.osr_h = BME280_OVERSAMPLING_1X;
    dev->settings.osr_p = BME280_OVERSAMPLING_16X;
    dev->settings.osr_t = BME280_OVERSAMPLING_2X;
    dev->settings.filter = BME280_FILTER_COEFF_16;

    settings_sel = BME280_OSR_PRESS_SEL | BME280_OSR_TEMP_SEL | BME280_OSR_HUM_SEL | BME280_FILTER_SEL;

    rslt = bme280_set_sensor_settings(settings_sel, dev);

    NRF_LOG_INFO("Temperature, Pressure, Humidity\r\n");
    /* Continuously stream sensor data */
    while (1) {
        rslt = bme280_set_sensor_mode(BME280_FORCED_MODE, dev);
        /* Wait for the measurement to complete and print data @25Hz */
        dev->delay_ms(40);
        rslt = bme280_get_sensor_data(BME280_ALL, &comp_data, dev);
        print_sensor_data(&comp_data);
    }
    return rslt;
}

void print_sensor_data(struct bme280_data *comp_data)
{
#ifdef BME280_FLOAT_ENABLE
        printf("%0.2f, %0.2f, %0.2f\r\n",comp_data->temperature, comp_data->pressure, comp_data->humidity);
#else
        NRF_LOG_INFO("%ld, %ld, %ld\r\n",comp_data->temperature, comp_data->pressure, comp_data->humidity);
#endif
}

/* End own function declarations and definitions */

int main(void)
{   
    APP_ERROR_CHECK(NRF_LOG_INIT(NULL));
    NRF_LOG_DEFAULT_BACKENDS_INIT();

    nrf_drv_spi_config_t spi_config = NRF_DRV_SPI_DEFAULT_CONFIG;
    //spi_config.ss_pin   = SPI_SS_PIN;
    spi_config.ss_pin   = NRF_DRV_SPI_PIN_NOT_USED; /* Initialize p_spi_instance without an ss_pin. ss_pin = NRF_DRV_SPI_PIN_NOT_USED*/	 
    spi_config.miso_pin = SPI_MISO_PIN;
    spi_config.mosi_pin = SPI_MOSI_PIN;
    spi_config.sck_pin  = SPI_SCK_PIN;
    //APP_ERROR_CHECK(nrf_drv_spi_init(&spi, &spi_config, spi_event_handler, NULL)); //Initialize in non blocking mode
    APP_ERROR_CHECK(nrf_drv_spi_init(&spi, &spi_config, NULL, NULL)); // Initialize in blocking mode

    NRF_LOG_INFO("SPI example started.");
    /* End example code */
    
    /* Own code */
    //APP_ERROR_CHECK(nrf_drv_spi_init(&spi, &spi_config, NULL, NULL)); // Initialize in blocking mode
    struct bme280_dev dev;
    int8_t rslt = BME280_OK;

    /* Sensor_0 interface over SPI with native chip select line */
    dev.dev_id = 0;
    dev.intf = BME280_SPI_INTF;
    dev.read = spi1_read_transfer;
    dev.write = spi1_write_transfer;
    dev.delay_ms = user_delay_ms;

    rslt = bme280_init(&dev);
    stream_sensor_data_forced_mode(&dev);
}

ma-harvey, 2018-12-21:
BME280 uses automatic protocol detection, based on the status of the CSB pin, as described in paragraph 6.1. 

Try forcing the part to be in SPI mode before trying to read Chip ID by pulling low the CSB line (even without communication) for at least 40ns. (t_setup + t_hold)

Alternatively, if it is easier to program, simply try to write any value to address 0xD0 (Chip ID) and ignore the error message if the transaction fails. Since this register it read-only, the part will ignore the write command.

ma-harvey, 2018-12-21:
What I means by that, is to pull it low, then back high so that the part is already in SPI mode before the real SPI transaction.

An initial dummy transaction could achieve that.

ASL07, 2018-12-21:
@ma-harvey  Thanks for your quick replies.